innoveit/react-native-ble-manager #927: How often should I be receiving notications with the BleManagerDidUpdateValueForCharacteristic event listener callback?

thomasbing4589, 2022-11-21:
I've used this library to implement BLE communications between my app and a BLE peripheral with +-5 characteristics all sending notifications at different intervals (some of which are very fast, like every 30-50ms). I'm seeing poor performance when the peripheral is sending too many notifications in a short space of time. What kind of interval shoud I be expecting data to come through? How can I improve the app's callback to be able to handle more data at a higher speed?

I can include code snippets if necessary, but I believe this isn't a code-specific question yet.

Maxoos, 2022-11-23:
I ask myself the same question, I'm looking at putting BLE on a seperate process with [this](https://github.com/mrousavy/react-native-multithreading).

One thing I've noticed is when you compile the app to production the performance is better, prob due to all the logs the lib spits up.

I would say you should set updates as fast as you can but try to keep it isolated and not refresh the interface unless needed. Use Memo and tricks like that.

thomasbing4589, 2022-11-24:
First of all, thank you so much for your advice. It's a great help.

> One thing I've noticed is when you compile the app to production the performance is better, prob due to all the logs the lib spits up. 

I also noticed this, but it's still not able to handle very frequent notifications. Do you have any ideas for how to improve the efficiency of the callback? 

> I would say you should set updates as fast as you can but try to keep it isolated and not refresh the interface unless needed. Use Memo and tricks like that.

Yeah at the moment I don't think the refreshing of the interface is causing this as it's just writing to state in a React context.

Maxoos, 2022-11-24:
I would try putting all the BLE handling logic on a different process, as well as processing and only pass what you need to the react native process. You can also try a new headless JS process and run it there, use events to communicate messages.  I think I'm about to try something like this as well, because I can fill that the app is slowing down when getting many notifications.

closed 2023-12-04 by marcosinigaglia