innoveit/react-native-ble-manager #931: Wrong connectGatt version used for Android M and Android N (?)

Marvedog, 2022-12-05:
**Describe the bug**
This is primarily a question that needs clarification as I am not that familiar with Android Bluetooth internals.

According to the Android docs, the `conenctGatt` method used here

https://github.com/innoveit/react-native-ble-manager/blob/master/android/src/main/java/it/innove/Peripheral.java#L109

Was made available in API level 26.

https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback,%20int,%20int,%20android.os.Handler)

Can it be clarified why it is used for Android M and above? I am experiencing connectivity issues when using an Android N device. The exact same implementation works fine. 

**To Reproduce**
1. Apply fix in #916 
2. Attempt to connect w/ Android N device

**Expected behavior**
A proper connection.

**Smartphone (please complete the following information):**
 - Device: Android Nougat
 - OS: Android Nougat (7.1.1)
 - react-native-ble-manager version: master 
 - react-native version: 0.68.x

**Additional context**
N/A

marcosinigaglia, 2022-12-14:
Hi @Marvedog , I'm not expert too, I checked with Android Studio and it also suggest  the  `>= Build.VERSION_CODES.M` check. I think the the correct doc reference is that: https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback,%20int)

Marvedog, 2022-12-14:
I am able to make it work with the current implementation, but it is weird that the documentation suggests this.

closed 2023-12-07 by marcosinigaglia