innoveit/react-native-ble-manager #939: Blemanager.connect always return "Connection Error" in very first time connection.

RuchiraSwarnapriya, 2023-01-04:
**Before open an issue**
- Check the closed issues, your question maybe is not new.
- We can't debug or reproduce hardware issue.
- If the library is not working 99% you did something wrong, in the code, installation or in the phone permissions.

**Describe the bug**
When I call the Blemanager.connect() in very first time it always return a "Connection Error". After I call the method again in second time it's working smooth without any issue.

**To Reproduce**
My Implementation,

```
const callDelayed = (call, delay = 300) => {
  return new Promise((resolve, reject) => {
    setTimeout(() => {
      call().then(resolve, reject);
    }, delay);
  });
};

const value = await callDelayed(() =>
        BleManager.connect(macAddressOrUUID)
          .then(
            () => true,
          )
          .catch((error) => {
            console.log('failed to connect ', error); //in very first time this catch triggers.
            return false;
          }),
 );
```

**Expected behavior**
Connect without any error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Samsung A33
 - Android 12
 - react-native-ble-manager version: 8.4.4
 - react-native version: 0.66.5

RuchiraSwarnapriya, 2023-01-05:
@marcosinigaglia  can you take look

samuel-gallet, 2023-01-16:
I got the same behaviour on Android. It's working great on iOS

traianpopescu95, 2023-03-23:
I have the same problem on Android. 

Same code used to work perfectly two months ago. 
iOS also working perfectly. I also noticed this adb log today, maybe it helps. 
The command I used is `adb logcat "*:E`

Any further assistance would be appreciated.

traianpopescu95, 2023-04-01:
For anyone experiencing this, I managed to find a workaround that is working on **some** Android devices.

This is kinda ugly but at least it works, waiting for a better fix for this. 

Here is my workaround:

 ```
const connectToBluetooth = async (peripheral: Peripheral) => {
        let success = false;
        for (let i = 0; i < 20 && !success; i++) {
            try {
                await BleManager.connect(peripheral.id)
                    .then(async () => {
                        success = true;
                        console.log("Ble Manager connected !")
                        setTimeout(async () => {
                            await handleBluetoothConnection(peripheral);
                        }, 1000);
                    })
            } catch (err) {
                console.log(`Attempting to connect (${i + 1}/20): ${JSON.stringify(err)}`);
                await new Promise(resolve => setTimeout(resolve, 1000));
            }
        }
        if (!success) {
            console.log("Failed to connect after 20 attempts.");
        }
    }

```

KonstantinKostianytsia, 2023-06-04:
It was the same problem for me. I resolved it by using react-native-ble-plx connectToDevice method. Maybe, the solution is using autoConnect parameter because I could not connect to my BLE device without it.
```
const services = await this.manager.connectToDevice(id, {
      autoConnect: true,
});
```

ZenAl3x, 2023-06-13:
> It was the same problem for me. I resolved it by using react-native-ble-plx connectToDevice method. Maybe, the solution is using autoConnect parameter because I could not connect to my BLE device [...]

Is the rest of your code from ble-manager and working? Or did you convert everything to ble-plx?

KonstantinKostianytsia, 2023-06-13:
> Is the rest of your code from ble-manager and working? Or did you convert everything to ble-plx?

I also used `cancelDeviceConnection`, 'writeCharacteristicWithResponseForDevice' from react-native-ble-plx because I also faced issues with writing with ble-manager. I think I will also use ble-plx for reading data. I use ble-manager to init BLE manager, enable Bluetooth, scan, and get discovered devices. So I just rewrote functions that didn't work and this combination of libraries works fine.
I attached the interface of my service.
[image]

ZenAl3x, 2023-06-13:
Amazing, thank you for you help!

marcosinigaglia, 2023-06-13:
> It was the same problem for me. I resolved it by using react-native-ble-plx connectToDevice method. Maybe, the solution is using autoConnect parameter because I could not connect to my BLE device [...]

Now the library has the autoconnect flag, can you try if it works?

rakshitbharat, 2023-11-22:
global.isBLEManagerInitialized = false;
if (!global.isBLEManagerInitialized) {
  BleManager.start({showAlert: false}).then(() => {
    console.error('Module initialized');
    global.isBLEManagerInitialized = true;
  });
}

add this in index.js and make sure BleManager.start will never be called again anywhere in app

jtovarto, 2023-11-22:
> Now the library has the autoconnect flag, can you try if it works?

Normally on iOS the auto connect feature works fine without using the flag, but on android it is required. For me both platforms works fine with the flag enabled.

closed 2023-12-05 by marcosinigaglia