innoveit/react-native-ble-manager #942: Android Application crash

danbITC, 2023-01-10:
**Describe the bug**
The BLE Manager crashes when connect is called.
I was able to insert some log statements and it looks like its not actually executing the code that is being posted to the mainHandler in Peripheral.java, but the connect function itself is being called correctly.

**To Reproduce**
Build Environment
JVM: 11.0.16.1 (JDK is the same)
Groovy: 2.4.15
Kotlin: 1.2.61
Gradle: 4.10.1
React Native: 0.63.5
Gradle (Android Plugin): 3.3.0

Steps to reproduce the behavior:
1. git clone v8.6.0 of the repository
2. Integrate into an Application with Bluetooth for testing
3. Attempt to connect to the device
4. Application crashes with the following stacktrace

`01-10 16:52:56.691 21532 21608 D ReactNativeBleManager: start
01-10 16:52:56.693 21532 21608 D ReactNativeBleManager: BleManager initialized
01-10 16:52:56.709 21532 21608 D ReactNativeBleManager: checkState
01-10 16:52:56.710 21532 21608 D ReactNativeBleManager: state:on
01-10 16:52:56.769 21532 21608 D ReactNativeBleManager: Get connected peripherals
01-10 16:52:56.777 21532 21608 D ReactNativeBleManager: Connect to: 94:54:93:46:E5:5E`

**Expected behavior**
The BT device should be connected and running in the main thread

**Smartphone (please complete the following information):**
 - Device: Pixel 6 Pro and Samsung S20+ 5G
 - OS: Android 12 & Android 13
 - react-native-ble-manager version: 8.6.0 
 - react-native version: 0.63.5

I was able to compile without issues with both 8.5.0 and 8.6.0, but only 8.5.0 was functional.

marcosinigaglia, 2023-01-11:
Hi @danbITC , I think is a permission problem.

danbITC, 2023-01-11:
Hi @marcosinigaglia, 
I am able to scan for devices without issue and it is only upon connecting that it crashes.

The following are the relevant permissions inside the application AndroidManifest.xml
`<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />`
`<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />`

The following permissions are within the BleManager AndroidManifest.xml
`    <uses-permission android:name="android.permission.BLUETOOTH" />`
`    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />`
`    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />`

The build.gradle is targeting the following SDK.

`buildscript {

    ext {
        buildToolsVersion = "30.0.3"
        minSdkVersion = 26
        compileSdkVersion = 30
        targetSdkVersion = 30
        supportLibVersion = "28.0.0"
    }`

I have tried changing around the target SDK as well as trying various different permissions within the AndroidManifest.xml and I was not able to connect still.

Adding the following permissions from the readme caused an immediate application crash upon opening.

` <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />`
    `<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />`

`<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28"/>`
    `<uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION" tools:targetApi="Q"/>`

Please let me know if you need further information.

marcosinigaglia, 2023-01-11:
You are missing the "android.permission.BLUETOOTH_CONNECT" permission.

danbITC, 2023-01-11:
I created a build with the addition of the following permission inside the application AndroidManifest.xml

`	<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />`

The result was that the application crashed immediately upon launch.

I also ran a build with the following permissions inside the application AndroidManifest.xml

`<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" `/>
`<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />`

This also crashed upon launch.

gorhgorh, 2023-01-17:
it may be related to a change in android 12, now connect is a runtime permission, so you need to ask for user consent before initialising the BLE lib

doc here may help https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

here is a sample of the function i use to grant this permission before calling any of the ble function, I have to do this for scan as well

``` js
const requestConnectPermission = async (valPerm) => {
  try {
    const granted = await PermissionsAndroid.request(
      'android.permission.BLUETOOTH_CONNECT',
      {
        title: 'Ble connect permissions',
        message:
          'this app need to be able to connect to peripherals',
        buttonNeutral: "Ask Me Later (app won't work)",
        buttonNegative: 'Cancel',
        buttonPositive: 'OK'
      }
    )

    if (granted === PermissionsAndroid.RESULTS.GRANTED) {
      console.info('You can use connect to devices')
    } else {
      console.info('Connect permission denied')
    }
    if (valPerm) valPerm(true)
  } catch (err) {
    console.info(err)
  }
}
```

ablack34, 2023-02-21:
@danbITC did @gorhgorh solution above work for you?

danbITC, 2023-02-21:
Unfortunately no, I spent some time trying to debug this last week and I was able to get the application to the point of scanning, but now it fails on connect.

Since our application uses 0.63.5 for react native I had to switch to a different permissions management package instead of using the built-in one. 

I'm not sure if its an issue with my manifest file or something else. I have not had time to dedicate to resolving this issue, but for now we are using 8.6.0 without issue.

niubiletei, 2023-02-28:
I have the same problem as you,but when I use 8.5.0 for ble-manager,our android application still crash,Can I refer to your AndroidManifest.xml file?

niubiletei, 2023-03-01:
@danbITC Please help me！

niubiletei, 2023-03-01:
After testing，I'm locking the problem to android12 or later versions,do you have good solutions? @marcosinigaglia

marcosinigaglia, 2023-03-01:
@niubiletei is not a library issue, probably you are try to scan or connect without the permission.

closed 2023-03-01 by marcosinigaglia

niubiletei, 2023-03-02:
[image]
@marcosinigaglia maybe this is a runtime problem,I try to requst he following three permissions and successfully solved the problem！
[image]

niubiletei, 2023-03-02:
@marcosinigaglia  Maybe this is a runtime problem,I try to request the following three permissions about android 12 and the problem was successfully solved!
[image]
[image]