innoveit/react-native-ble-manager #943: support Android 13 BLE library updates

matgardon, 2023-01-16:
Hi, as you may be aware, android 13 partially switched it's BLE stack to [Gabeldorsche](https://www.xda-developers.com/android-13-gabeldorsche-bluetooth-stack/) this summer. This change mainly impacts Advertising & Scanning for now.

As a consequence, several methods used are now flagged as deprecated, and some FLAGS have been added (for example in scan results error cases).

changes can be found here:
- https://developer.android.com/sdk/api_diff/33/changes/pkg_android.bluetooth
- https://developer.android.com/sdk/api_diff/33/changes/pkg_android.bluetooth.le

I think it would be wise to switch to the non-deprecated methods when possible (ie. on Android >=13) because some of those deprecated methods had race conditions / thread safety issues as [discussed here](https://github.com/NordicSemiconductor/Android-BLE-Library/pull/425). It could also be a complementary fix to the recently merged PR #935.

A good summary of the changes needed might be found in the Nordic lib in the following 2 PRs (both included in latest release, 2.6.0) :
- https://github.com/NordicSemiconductor/Android-BLE-Library/pull/424
- https://github.com/NordicSemiconductor/Android-BLE-Library/pull/425

I don't have much time right now but if I can help on that integration, I will.

Also, not directly related, but be aware that the new android 13 scan introduced issues which consequences I'm not yet 100% certain of, but they include:

- scan returning only partial data when manufacturing data include padding 0s [as documented here](https://issuetracker.google.com/issues/246824348), for which a fix was provided by google but not yet deployed to all android 13 devices (samsung devices migrated to tiramisu seem to all have the problem as of today from my own tests).

- a seemingly unstable scan behavior of this lib, not respecting the scan duration (scan indefinitely), nor respecting `stopScan` command. It may be related to a new behavior of the underlying android lib with reportDelay=0 causing too much spam on the `ScanCallback`, and so this lib might have to adjust to the new performance improvements ? still under investigation on my side.

roysG, 2023-07-04:
Hi,
I see that there is no any updated in the repository. how should i get the new fix?

marcosinigaglia, 2023-07-04:
The library is working also on Android 13

askkerr, 2023-08-30:
I'm not sure that it is. Android 13, Samsung Galaxy S21 Ultra. No peripherals added during scan

matgardon, 2023-08-30:
It is working, this issue is about supporting new non-deprecated methods for certain inner workings as explained in the issue desc. This is not (yet) a blocking issue. If you have issues with scan, there are probably a whole lot of reasons why, and they are probably not related to this library as scanning conditions may be tricky when first putting your hands into BLE on android.

MuhammadFaisal215, 2023-11-16:
Hi,
I am facing the same issue. When i use my App in android 11 it works fine and scan all nearby bluetooth devices, including classical and BLE devices. But when i use my App with android 12, 13, and 14, my app only scans classical bluetooth devices like, headphone, mobiles, smart watches, but not my BLE devices.
Can anyone help me on this.
Thanks.

askkerr, 2023-11-16:
> Hi, I am facing the same issue. When i use my App in android 11 it works fine and scan all nearby bluetooth devices, including classical and BLE devices. But when i use my App with android 12, 13, and 14, my app only scans classical bluetooth devices like, headphone, mobiles, smart watches, but not my BLE devices. I have requested and granted all type of permissions required regarding bluetooth and location, but didn't work. Can anyone help me on this. Thanks.

Make sure in your manifest you have
`    <uses-feature
        android:name="android.hardware.bluetooth_le"
        android:required="true" />

<uses-permission
        android:name="android.permission.BLUETOOTH_SCAN"
        android:usesPermissionFlags="neverForLocation" />
 <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
 <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
 <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
`
With the "neverforlocation" tag on the BLUETOOTH_SCAN permission, it will allow BLE scanning without needing location permissions. Do keep in mind even if permissions are granted, the respective adapters must be on for the scan to work.

MuhammadFaisal215, 2023-11-16:
> > Hi, I am facing the same issue. When i use my App in android 11 it works fine and scan all nearby bluetooth devices, including classical and BLE devices. But when i use my App with android 12, 13, and 14, my app only scans classical bluetooth devices like, headphone, mobiles, smart watches, but not my BLE devices. I have requested and granted all type of permissions required regarding bluetooth and location, but didn't work. Can anyone help me on this. Thanks.
> 
> Make sure in your manifest you have uses-feature android.hardware.bluetooth_le required="true" android.permission.BLUETOOTH_SCAN "neverForLocation" android.permission.BLUETOOTH_ADMIN android.permission.BLUETOOTH_CONNECT android.permission.ACCESS_FINE_LOCATION
> 
> With the "neverforlocation" tag on the BLUETOOTH_SCAN permission, it will allow BLE scanning without needing location permissions. Do keep in mind even if permissions are granted, the respective adapters must be on for the scan to work.

Thanks @askkerr.
I have already added the permissions in manifest file that you have listed. And also I have granted all the permissions. My adapters are on. But still getting same issue. Not scanning BLE devices, but scanning other bluetooth devices.

askkerr, 2023-11-16:
> Thanks @askkerr.
> I have already added the permissions in manifest file that you have listed. And also I have granted all the permissions. My adapters are on. But still getting same issue. Not [...]

I'd love to help, but I'm not really even sure how you are finding classic devices. This library should only be for BLE.

askkerr, 2023-11-17:
> Thanks @askkerr. I have already added the permissions in manifest file that you have listed. And also I have granted all the permissions. My adapters are on. But still getting same issue. Not [...]

@MuhammadFaisal215 just a thought, but are you specifying the service UUIDs of the devices you plan to be searching for? In my experience Android 13 on a Samsung S21 Ultra did not care if I specified service UUIDs, but we had issues on iOS if we didn't specify any UUIDs. Maybe it is a similar case with different hardware?

C0zmaCatalin, 2023-12-11:
> Hi, I am facing the same issue. When i use my App in android 11 it works fine and scan all nearby bluetooth devices, including classical and BLE devices. But when i use my App with android 12, 13, [...]

Same for me... I couldn't find any solution to fix this :(

Aladin-83, 2023-12-17:
Good morning,
Being stuck on the same problem, after a few hours of testing and research, I found the solution! It is imperative to put 
```
`<uses-permission android:name="android.permission.BLUETOOTH_SCAN"
         android:usesPermissionFlags="neverForLocation" >

  in the manifest and check the authorizations in the program:

if (Platform.OS === 'android' && Platform.Version >= 23) {
             console.log('In verifyBluetoothAuthorization...',
                             Platform.Version)
             PermissionsAndroid.requestMultiple([
                 PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
                 PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
                 PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
                
               ]).then(result => {
                
                 if (
                   (result['android.permission.BLUETOOTH_SCAN'] &&
                   result['android.permission.BLUETOOTH_CONNECT'] &&
                   result['android.permission.ACCESS_FINE_LOCATION'] === 'granted')
                   ||
                   (result['android.permission.BLUETOOTH_SCAN'] &&
                   result['android.permission.BLUETOOTH_CONNECT'] &&
                   result['android.permission.ACCESS_FINE_LOCATION'] === 'never_ask_again')
                 ) {
                   console.log('User accepted');
                   ActiveBluetoothInDevice()
                 } else {
                   console.log('User refused'); }
               });
           } ```

and suddenly everything works!

usama-tariq-qbatch, 2024-02-29:
> Good morning, Being stuck on the same problem, after a few hours of testing and research, I found the solution! It is imperative to put
>
> ```
> `<uses-permission [...]

Thanks alot @Aladin-83 I was searching for the fix for a long time, and your solution worked magically.

Aladin-83, 2024-03-03:
Bonjour,

Ravi pour vous !

Cordialement

P. Nolin

lucaswitch, 2024-03-27:
So it's not a library issue as the solution is a permission issue? Never had this problems on Android 13 and higher.

closed 2024-12-16 by marcosinigaglia