innoveit/react-native-ble-manager #951: SERVICE UUID AND CHARACTERISTIC UUID

mohsinmughal1630, 2023-01-31:
const SERVICE_UUID = "0000FFE0-0000-1000-8000-00805F9B34FB";
const CHARACTER_UUID = "0000FFE1-0000-1000-8000-00805F9B34FB";

That is my device service uuid and character uuid which i get from device developer but from when this library i send data getting this error "**Error writing 0000FFE1-0000-1000-8000-00805F9B34FB status = 10**"

Please help me what i doing wrong my code

  BleManager.retrieveServices(props.route.params.bluetoothData.id).then(
      (peripheralData) => {
        const mData = `${firstName} \n${lastName}`

        const data = stringToBytes(mData);
            BleManager.write(
                props.route.params.bluetoothData.id,
                SERVICE_UUID,
                CHARACTER_UUID,
                data
              )
                .then(() => {
                  // Success code
                  console.log("Write: " + data);
                })
                .catch((error) => {
               
                  console.log(error);
                });
        }
      
      
    ).catch((err)=>{
        dispatch(startLoader(false));
        Alert.alert("Error!", err)
    });

mohsinmughal1630, 2023-01-31:
Please help me with this humble request

mohsinmughal1630, 2023-02-05:
???

sefam-stoussaint, 2023-02-07:
Hello,
I think your UUIDs should be "FFE0" and "FFE1"

aje, 2023-07-14:
Your data is wrong, it should be like: 

```
import { Buffer } from 'buffer';
const buffer = Buffer.from('test');
```

closed 2023-12-05 by marcosinigaglia