innoveit/react-native-ble-manager #967: Advertising does not hold serviceUUIDs

medonja, 2023-02-26:
**Describe the bug**
My advertisement does not hold a serviceUUIDs for any of the found Bluetooth devices returned in the BleManagerDiscoverPeripheral handler. 
This is how a typical response looks like:

` {"advertising": {"isConnectable": true, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "AgEaC/9MAAkGAzHAqACzAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}}, "id": "13:B2:6B:EA:63:B1", "name": null, "rssi": -53}
`

**Expected behavior**
I expected the advertising to have the exact attributes as describe here: https://github.com/innoveit/react-native-ble-manager#blemanagerdiscoverperipheral
It seems also that some attributes have a slightly different name than described in the documentation.,

**Smartphone (please complete the following information):**
 - Device: Google Pixel 6 (but also a somewhat older Android device such as the Samsung A50)
 - OS: Android 13
 - react-native-ble-manager version: 8.8.0
 - react-native version: 0.68.2

tobiyas09, 2023-03-14:
I have the same problem on v8.8.0, but if you try to run the example on the master branch it returns serviceUUIDs. 
Maybe they fixed this and we will get this with the next release?

medonja, 2023-03-14:
No idea, but I resorted to react-native-ble-plx meanwhile... a pity.

marcosinigaglia, 2023-03-15:
Hi, you need to call retrieveServices to have the service UUIDS.

closed 2023-03-15 by marcosinigaglia

tobiyas09, 2023-03-15:
> Hi, you need to call retrieveServices to have the service UUIDS.

Hi, I don't think that was the problem, scan results didn't have serviceUUIDs in AdvertisingData. The RetrieveServices method needs to be called before writing, reading or starting a notification. 
What fixed it for me was switching forceLegacy to  **false** in the start method.

mokaty1818795, 2023-11-02:
> I have the same problem on v8.8.0, but if you try to run the example on the master branch it returns serviceUUIDs. Maybe they fixed this and we will get this with the next release?

did you get the solution ?

tobiyas09, 2023-11-02:
> did you get the solution ?

Yes, I put forceLegacy to false in the start method

mokaty1818795, 2023-11-02:
> > did you get the solution ?
> 
> Yes, I put forceLegacy to false in the start method

this is my code snipped
import React, {
  ReactNode,
  createContext,
  useContext,
  useEffect,
  useState,
} from 'react';
import {NativeEventEmitter, NativeModules} from 'react-native';
import BleManager, {
  BleScanCallbackType,
  BleScanMatchMode,
  BleScanMode,
  Peripheral,
  PeripheralInfo,
} from 'react-native-ble-manager';
import {Buffer} from 'buffer';
import {BluetoothEventTypes} from './event-types';
import {handleAndroidPermissions} from '../utils/bluetooth-permission';

declare module 'react-native-ble-manager' {
  interface Peripheral {
    connected?: boolean;
    connecting?: boolean;
  }
}

type BluetoothContextType = {
  devices: Map<string, Peripheral>;
  scanning: boolean;
  scanDevices: () => void;
  connectDevice: (id: string) => Promise<void>;
  connectedDevice: PeripheralInfo | null;
  disconnect: () => void;
  printContent: (input: string) => Promise<void>;
};

type BluetoothProviderType = {
  children: ReactNode;
};

const bluetoothManagerModule = NativeModules.BleManager;
const bluetoothManagerEmitter = new NativeEventEmitter(bluetoothManagerModule);

const SECONDS_TO_SCAN_FOR = 5;
const SERVICE_UUIDS: string[] = [];
const ALLOW_DUPLICATES = true;

const BluetoothContext = createContext({} as BluetoothContextType);

const BluetoothProvider = ({children}: BluetoothProviderType) => {
  const [scanning, setScaning] = useState(false);
  const [peripherals, setPeripherals] = useState(
    new Map<Peripheral['id'], Peripheral>(),
  );
  const [connectedDevice, setConnectedDevice] = useState<PeripheralInfo | null>(
    null,
  );

  const addOrUpdatePeripheral = (id: string, updatedPeripheral: Peripheral) => {
    setPeripherals(map => new Map(map.set(id, updatedPeripheral)));
  };

  const handleRetrieveConnecteds = async () => {
    const connectDevices = await BleManager.getConnectedPeripherals();
    if (connectDevices.length === 0) {
[... 137 lines cut ...]
    };
  });

  return (
    <BluetoothContext.Provider
      value={{
        devices: peripherals,
        scanning,
        scanDevices,
        connectDevice,
        connectedDevice,
        disconnect,
        printContent,
      }}>
      {children}
    </BluetoothContext.Provider>
  );
};

function useBluetooth() {
  const context = useContext(BluetoothContext);
  return context;
}

export {BluetoothProvider, useBluetooth};

mokaty1818795, 2023-11-02:
> > did you get the solution ?
> 
> Yes, I put forceLegacy to false in the start method

i did and still get the same issue