innoveit/react-native-ble-manager #985: BleManager.disconnect not working on Android

hzcfy, 2023-03-20:
**Before open an issue**
- Check the closed issues, your question maybe is not new.
- We can't debug or reproduce hardware issue.
- If the library is not working 99% you did something wrong, in the code, installation or in the phone permissions.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - react-native-ble-manager version:  8.8.0
 - react-native version: 0.71.3

**Additional context**
Add any other context about the problem here.

schimini, 2023-04-05:
@hzcfy can you provide any context and update the issue description?

hzcfy, 2023-04-05:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

schimini, 2023-04-05:
Ahah I've translated this and this is an automatic email response. Hope you have a nice vacation!

closed 2023-12-05 by marcosinigaglia