innoveit/react-native-ble-manager #994: Cannot use readDescriptor

rishiankush, 2023-03-31:
I am getting this error while using `BleManager.readDescriptor`
`TypeError: _reactNativeBleManager.default.readDescriptor is not a function`

Here is my code:
```
const connect = async (id: string, navigation: any) => {
    console.log('[navigation connect method]', navigation)

    console.debug('[connectBT]', id);
    connetedDevice('');
    setConnectionError('');
    setConnectState(LOADING_STATE.init);
    await BleManager.connect(id)
      .then(() => {
        // Success code
        connetedDevice(id);
        setConnectionError('');
        setConnectState(LOADING_STATE.succuss);
        console.debug('[connectBT] Connected');
        BleManager.retrieveServices(id).then(
            async (peripheralInfo) => {
              // Success code
              console.log("[peripheralInfo]", peripheralInfo);
              if (peripheralInfo.characteristics) {
                for (let characteristic of peripheralInfo.characteristics) {
                  if (characteristic.descriptors) {
                    for (let descriptor of characteristic.descriptors) {
                      try {
                        let data = await BleManager.readDescriptor(peripheralInfo.id, characteristic.service, characteristic.characteristic, descriptor.uuid);
                        console.debug(`[connectPeripheral][${peripheralInfo.id}] descriptor read as:`, data);
                        setTimeout(() => {
                          navigation.navigate('BluetoothDeviceDetail', {peripheralInfo})
                        }, 3000);
                      } catch (error) {
                        console.error(`[connectPeripheral][${peripheralInfo.id}] failed to retrieve descriptor ${descriptor} for characteristic ${characteristic}:`, error);
                      }
                    }
                  }
                }
              }
            }
        );
      })
      .catch((error) => {
        // Failure code
        connetedDevice('');
        setConnectionError(error);
        setConnectState(LOADING_STATE.error);
        console.error('[connectBT] Error ', error);
      });
  };
```

marcosinigaglia, 2023-03-31:
Hi @rishiankush , I published now the new version with readDescriptor function.

closed 2023-03-31 by marcosinigaglia

rishiankush, 2023-04-04:
Working now!! perfect Thanks! @marcosinigaglia