innoveit/react-native-ble-manager #996: readDescriptor causing app to crash

ryd0rz, 2023-04-05:
Getting this when using readDescriptor on iOS:

2023-04-05 10:59:37.123806-0400 app[19407:1957904] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFNumber length]: unrecognized selector sent to instance 0xaf9c9b859f6bed2a'
*** First throw call stack:
(0x1a7c79d78 0x1c08de734 0x1a7d57b88 0x1a7c0e988 0x1a7c0da60 0x100b37ef4 0x1c644364c 0x1c644389c 0x1c643f994 0x1c6429af0 0x1c646a2b8 0x1c64596c0 0x10366cc6c 0x10366e7bc 0x1036768a4 0x103677614 0x10367eac8 0x10367e7a4 0x1a7c322f0 0x1a7bec1f4 0x1a7bff6b8 0x1c3c99374 0x1aa564e88 0x1aa2e65ec 0x100139bec 0x103299ce4)
libc++abi: terminating with uncaught exception of type NSException
dyld4 config: DYLD_LIBRARY_PATH=/usr/lib/system/introspection DYLD_INSERT_LIBRARIES=/Developer/usr/lib/libBacktraceRecording.dylib:/Developer/usr/lib/libMainThreadChecker.dylib:/Developer/Library/PrivateFrameworks/DTDDISupport.framework/libViewDebuggerSupport.dylib
terminating with uncaught exception of type NSException
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFNumber length]: unrecognized selector sent to instance 0xaf9c9b859f6bed2a'

I'm using the example code. Fails right on the line: 

`let data = await BleManager.readDescriptor(peripheral.id, characteristic.service, characteristic.characteristic, descriptor.uuid);

I'm assuming we don't really need this? It's just to get more information about the characteristic... would be nice to have though..

ryd0rz, 2023-04-10:
> Hi, did you found any solutions? I am facing the same problem.

It's not supported in iOS I checked the exports and it's not there for iOS.

closed 2023-04-19 by ryd0rz

bilaltirmizi123, 2023-05-18:
@ryd0rz then how can we read descriptors data in iOS? please guide im stuck

ryd0rz, 2023-05-18:
@bilaltirmizi123 - If you must use iOS, I would advise using an app like "LightBlue" to read the descriptors for you.

bilaltirmizi123, 2023-05-18:
@ryd0rz I'm using LightBlue and BLE sensor app as well. I'm facing issue reading descriptors using BleManager.readDescriptor( ) function. its hanging my ios app.