innoveit/react-native-ble-manager #1013: Example app crashes when attempting to connect to a device

hypext1, 2023-05-09:
When I tap on a device to connect to it on the example app, the app just crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Download and compile example app
2. Attempt to connect to any BLE device
3. :(

**Expected behavior**
App connects to ble device without crashing

**Smartphone (please complete the following information):**
 - Device: iPhone 14 Pro
 - OS: ios 16.3.1
 - react-native-ble-manager version: 10.0.2
 - react-native version: 0.70.3

marcosinigaglia, 2023-05-09:
Check the error, probably is a permission problem.

closed 2023-05-09 by marcosinigaglia

hypext1, 2023-05-09:
> Check the error, probably is a permission problem.

This is the error, I don't think it's a permissions error:

2023-05-09 12:30:56.787656-0400 example[2022:83831] -[__NSCFNumber length]: unrecognized selector sent to instance 0xb4cc05ed40855beb
2023-05-09 12:30:56.788490-0400 example[2022:83831] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFNumber length]: unrecognized selector sent to instance 0xb4cc05ed40855beb'
*** First throw call stack:
(0x191afed94 0x18abb03d0 0x191c73b14 0x191b151c8 0x191b7b9f0 0x1057a60f4 0x1b2ad6d98 0x1b2ad6f5c 0x1b2ad3b40 0x1b2a9fb9c 0x1b2a9f71c 0x1b2a9f5f4 0x107ef4520 0x107ef6038 0x107efe0b0 0x107efee28 0x107f065f8 0x107f062dc 0x191b8dd18 0x191b6f650 0x191b744dc 0x1ccdd435c 0x193f0037c 0x193efffe0 0x104fafaec 0x1b0ffcdec)
libc++abi: terminating due to uncaught exception of type NSException

orlyprofili, 2023-08-14:
can confirm, is there a known workaround?

zvici, 2024-01-18:
me too
[image]

badpinkman, 2024-12-04:
How did you deal with it? I also encountered the problem that this program crashed and could not catch exceptions!!