innoveit/react-native-ble-manager #1014: Can't build library on iOS: 'React/RCTBridgeModule.h' file not found

joshhansen, 2023-05-16:
**Describe the bug**
When I build the iOS project in `/ios` on my Mac, it fails because it can't find `React/RCTBridgeModule.h`.

I would expect to see a `Podfile` to bring in the dependencies, but there is none, unlike in the example project in `/example` (which I am able to build). There is also no `README.md`, so I'm left without guidance on building the library itself.

This issue is slightly different from #211 - that issue regards projects that depend on react-native-ble-manager, while I'm trying to build the  library itself.

**To Reproduce**
`cd react-native-ble-manager/ios`
`xcodebuild`

The same error occurs when I do the build in XCode.

**Expected behavior**
There should be some means of installing the dependencies and completing the build, or at least guidance on how to do so.

**Screenshots**
n/a

**Environment:**
 - Device: some mac mini
 - OS: macOS 13.3.1
 - react-native-ble-manager version: current `master` (`6253b3e606219ecfce66cc8a07f89c10c3b707b5`)
 - react-native version: 0.71.0
 - XCode: 14.3

**Other:**
I am very new to React Native and ios development, so please forgive my ignorance if the solution is obvious.

joshhansen, 2023-05-17:
I did manage to build it - putting this here in case it helps. Steps:

1. `cd ios; pod init`

2. Replace the `Podfile` with this:
```require Pod::Executable.execute_command('node', ['-p',
  'require.resolve(
    "react-native/scripts/react_native_pods.rb",
    {paths: [process.argv[1]]},
  )', __dir__]).strip

require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'

platform :ios, min_ios_version_supported
prepare_react_native_project!

#linkage = ENV['USE_FRAMEWORKS']
#if linkage != nil
#  Pod::UI.puts "Configuring Pod with #{linkage}ally linked Frameworks".green
#  use_frameworks! :linkage => linkage.to_sym
#end

flipper_config = ENV['NO_FLIPPER'] == "1" ? FlipperConfiguration.disabled : FlipperConfiguration.enabled

target 'BleManager' do
  # Comment the next line if you don't want to use dynamic frameworks
  # use_frameworks!

  # Pods for BleManager

  config = use_native_modules!

  flags = get_default_flags()

  use_react_native!(
    :path => config[:reactNativePath],
    # Hermes is now enabled by default. Disable by setting this flag to false.
    :hermes_enabled => flags[:hermes_enabled],
    :fabric_enabled => flags[:fabric_enabled],
    # Enables Flipper.
    #
    # Note that if you have use_frameworks! enabled, Flipper will not work and
    # you should disable the next line.
    :flipper_configuration => flipper_config,
    # An absolute path to your application root.
    :app_path => "#{Pod::Config.instance.installation_root}/.."
  )

  post_install do |installer|
    # https://github.com/facebook/react-native/blob/main/scripts/react_native_pods.rb#L197-L202
    react_native_post_install(
      installer,
      config[:reactNativePath],
      :mac_catalyst_enabled => false
    )
    #__fix_double_definition_of_clockid_in_folly()
  end

end
```

3. `pod install`
4. (In my case at least) comment out the erroneous typedef in `Pods/Headers/Private/RCT-Folly/folly/portability/Time.h` (see facebook/flipper#834)
5. `xcodebuild -workspace ./BleManager.xcworkspace -scheme BleManager`

marcosinigaglia, 2023-05-18:
Hi @joshhansen , you need the Podfile in your project, not in the library.

joshhansen, 2023-05-18:
@marcosinigaglia I'm trying to build the BleManager.xcodeproj project in this repo, as I work on implementing peripheral mode features for this library. Running `xcodebuild` alone fails; but witih a Podfile in this repo I can build it.

Or could you recommend to me a different workflow for working on this library?

marcosinigaglia, 2023-05-19:
I never build or used the library without a react-native project because you can't test it without a real device, so I don't think what are you trying to do is useful.