innoveit/react-native-ble-manager #1016: iOS app stuck on readDescriptor function and Xcode showing Error

bilaltirmizi123, 2023-05-18:
iOS app stuck on readDescriptor function and Xcode showing following error: 

Thread 1: "-[__NSCFNumber length]: unrecognized selector sent to instance 0xb7bc8fec5ce713ae

Below is the code I'm pasting for reference: 

const connectPeripheral = async (peripheral: Peripheral) => {
    try {
      if (peripheral) {
        addOrUpdatePeripheral(peripheral.id, {...peripheral, connecting: true});

        await BleManager.connect(peripheral.id);
        console.debug(`[connectPeripheral][${peripheral.id}] connected.`);

        addOrUpdatePeripheral(peripheral.id, {
          ...peripheral,
          connecting: false,
          connected: true,
        });

        // before retrieving services, it is often a good idea to let bonding & connection finish properly
        await sleep(900);

        /* Test read current RSSI value, retrieve services first */
        const peripheralData = await BleManager.retrieveServices(peripheral.id);
        console.debug(
          `[connectPeripheral][${peripheral.id}] retrieved peripheral services`,
          peripheralData,
        );

        const rssi = await BleManager.readRSSI(peripheral.id);
        console.debug(
          `[connectPeripheral][${peripheral.id}] retrieved current RSSI value: ${rssi}.`,
        );

        if (peripheralData.characteristics) {
          for (let characteristic of peripheralData.characteristics) {
            if (characteristic.descriptors) {
              for (let descriptor of characteristic.descriptors) {

                try {
                  let data = await BleManager.readDescriptor(
                    peripheral.id,
                    characteristic.service,
                    characteristic.characteristic,
                    descriptor.uuid,
                  );
                  console.log(data)
                  console.debug(
                    `[connectPeripheral][${peripheral.id}] descriptor read as:`,
                    data,
                  );
                } catch (error) {
                  console.error(
                    `[connectPeripheral][${peripheral.id}] failed to retrieve descriptor ${descriptor} for characteristic ${characteristic}:`,
                    error,
                  );
                }
              }
            }
          }
        }

        // let p = peripherals.get(peripheral.id);
        // if (p) {
        //   addOrUpdatePeripheral(peripheral.id, {...peripheral, rssi});
        // }
      }
    } catch (error) {
      console.error(
        `[connectPeripheral][${peripheral.id}] connectPeripheral error`,
        error,
      );
    }
  };

hsponberg, 2023-06-18:
I believe this is caused by line 118 of `BleManager` https://github.com/innoveit/react-native-ble-manager/blob/master/ios/BleManager.m#L118

The code calls `[descriptor.value length]` which only works if the value is a certain type, like NSData, but the value is not guaranteed to be one of those types, eg. NSNumber.

birdlavv, 2023-08-24:
This may be related to https://github.com/innoveit/react-native-ble-manager/issues/1013. The issue is closed but is still active.

gharezlak, 2023-09-20:
I'm having this exact issue. Has anyone been able to solve this? Is there a way to ensure that the value being referenced by [descriptor.value length] is of a compatible datatype?

gharezlak, 2023-09-20:
Haha I just saw the comment about commenting out the nslog... I also just replaced that code block with 
```
if ([descriptor.value isKindOfClass:[NSData class]]) {
        NSLog(@"Read value [descriptor: %@, characteristic: %@]: (%lu) %@",
                    descriptor.UUID,
                    descriptor.characteristic.UUID,
                    [(NSData *)descriptor.value length],
                    descriptor.value);
} else {
        NSLog(@"Read value [descriptor: %@, characteristic: %@]: %@",
                    descriptor.UUID,
                    descriptor.characteristic.UUID,
                    descriptor.value);
}
```
and that worked too

gharezlak, 2023-09-21:
@marcosinigaglia submitted!

closed 2023-12-01 by marcosinigaglia