innoveit/react-native-ble-manager #1019: Scan sometimes don't find anything

TheMizuchi, 2023-05-26:
**Describe the bug**
When you scan, 1 time out of 6-10, the scan don't find anything and run out of times

**To Reproduce**
Steps to reproduce the behavior:
 - Take the example from the git and don't modify anything. 
 - Install the app on your smartphone and launch it
 - When the button Scan is available, press it.
 - Normally you will have a list of all BLE devices available nearby. But sometimes, it doesn't find anything.

**Expected behavior**
The method scan find all BLE devices available

**Smartphone (please complete the following information):**
 - Device: Redmi note 9S
 - OS: Android 13 (or MIUI 14)
 - react-native-ble-manager version: 9.3.1
 - react-native version: 0.70.5

**Additional context**
When you relaunch a scan with this app, the scan rework. I don't know why but it's work.

If it's happened to anyone else, a way to get around is to make the stop scan listener relaunch a scan if it doesn't find anything. The cons of that is the scan can be stuck for 3-15s and make the connection a bit long.

AzranAzwer, 2023-05-30:
Hi read this [Issue](https://github.com/innoveit/react-native-ble-manager/issues/922) May be help for you.

TheMizuchi, 2023-05-30:
Thanks but i have already do your solution. Maybe it is related to #943. If so, well, we need to wait for an update who can fix this problem

1mike12, 2023-06-15:
[image]
[image]

ic-twist, 2023-06-21:
@1mike12 Did you find out what the problem is?
Same issue on Android 13 Pixel 5a.
I'm using expo and installed `react-native-ble-manager` and `@matthewwarnes/react-native-ble-manager-plugin`.
Spined up example code with no change (besides to debug logs) and tried scanning - the scanning times out and no peripheral is found even tho there definitely are discoverable BLE devices nearby. 

Here's what I see in the console log:
[image]

Note the following:
1. `android.permission.BLUETOOTH_CONNECT` and `android.permission.BLUETOOTH_SCAN` permissions are both granted.
2. Scan promise returned successful almost immediately after scan is started.
3. Scan stops after timeout is reached.

So instead of scanning I just connected directly to the device using its MAC address. And that worked. (But I've read in some comments that I shouldn't be able to connect without scanning first?)

1mike12, 2023-06-21:
@ic-twist I actually didn't dig too much into it but I did run the app in the ble rich environment of an airplane and there, I got lots of rows to render, but I couldn't really tell if it was really working since every row seemed to be white text on white background. But I _think_ it was.

TheMizuchi, 2023-06-22:
@ic-twist Have you gave the Location permission? On android you must have access to the location to use bluetooth. Maybe it's just this. If not can we see your code?

TheMizuchi, 2023-06-22:
@1mike12 your images show exactly what i got on my side. When you're on the example app, you always see the "No peripheral" or is it just sometime like me?
(also sorry but i didnt understand your 2nd post xD)

ic-twist, 2023-06-22:
@TheMizuchi Granting the location permission did make a difference. But now when I start scanning, the bleManager is stuck on scanning and the promise is never resolved.
[image]

[image]

I'm using the example code that comes with the library, I just modified some console logs to show more information. But here's the code for scanning.

```
const SECONDS_TO_SCAN_FOR = 7;
const SERVICE_UUIDS: string[] = [];
const ALLOW_DUPLICATES = true;

  const startScan = () => {
    if (!isScanning) {
      // reset found peripherals before scan
      setPeripherals(new Map<Peripheral['id'], Peripheral>());

      try {
        console.debug('[startScan] starting scan...');
        setIsScanning(true);
        BleManager.scan(SERVICE_UUIDS, SECONDS_TO_SCAN_FOR, ALLOW_DUPLICATES, {
          matchMode: BleScanMatchMode.Sticky,
          scanMode: BleScanMode.LowLatency,
          callbackType: BleScanCallbackType.AllMatches,
        })
          .then(() => {
            console.debug('[startScan] scan promise returned successfully.');
          })
          .catch(err => {
            console.error('[startScan] ble scan returned in error', err);
          });
      } catch (error) {
        console.error('[startScan] ble scan error thrown', error);
      }
    }
  };
```

TheMizuchi, 2023-06-23:
Well for me all work on the example (except my problem of scanning...)
The Scan return almost immediately is normal, it's just start a scan thread. But normally the `handleDiscoverPeripheral` must be called when a peripheral is discover. I don't quite understand why but i don't think you issue is related to this issue. It's either a permission not granted (Coarse Location instead of Fine Location?) or your android manifest not well configured. Sorry for not being able to help so much ._.

sessions-matthew, 2023-06-25:
To get scanning to work for now I request precise location on Android 12. You may also need to also add it to your manifest.

``` js
PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION, '')
```

ic-twist, 2023-06-26:
I think I found out what my problem was:
1. I didn't request for `PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION` as @TheMizuchi and @sessions-matthew both mentioned. When the location permission is not granted, the BLE scan returns immediately without any device being found.
2. For some of my scans, I forgot to turn on Bluetooth on my phone. In this case, the scan does not return - like I described in [this comment](https://github.com/innoveit/react-native-ble-manager/issues/1019#issuecomment-1602642372)

Two suggestions/question I'm making:
1. Maybe we can add the location access request to the example code? Since the example code is not fully functional without it.
2. Maybe in the example code we can check for bluetooth's state and do not start the scan if the bluetooth is off - instead, notify the user?

1mike12, 2023-08-04:
So for me, the example works with just coarse "nearby devices" permission. I don't think that corresponds to `ACCESS_FINE_LOCATION`

Something that was kind of a duh moment was that I had Bluetooth turned off via the OS, but you can still hit scan and the example app chugs along. Not sure if this has anything to do with your guys' problems but worth checking. 

Also the latest example app I think someone must have changed the styling so now the names and text show up correctly. 

[image]

lucasmrau, 2023-09-06:
Guys, not returning nothing, could you share the piece of the code you are talking about? @1mike12 @ic-twist @sessions-matthew

ic-twist, 2023-09-06:
I propose adding this `enableBle` function to the example code.

```tsx
  function enableBle() {
    BleManager.enableBluetooth()
      .then(() => {
        // Success code
        console.log("The bluetooth is already enabled or the user confirmed");
      })
      .catch((error) => {
        // Failure code
        console.log("The user refuse to enable bluetooth");
      })
  }
```
And call the function after BLE starts
```tsx
      BleManager.start({ showAlert: false })
        .then(() => {
          console.debug('BleManager started.');
          enableBle();
        })
        .catch(error =>
          console.error('BeManager could not be started.', error),
        );
```

This can prevent the case when the user (like myself) forgot to turn on their bluetooth while trying the example code.

ReactRaylogic, 2023-10-17:
Hi, i am facing the same issue.
Already asked permission for ACCESS_FINE_LOCATION and BLUETOOTH_SCAN in the app.
When navigating to settings -> App permission 
I see that the app has Nearby device allowed.
When I click on allow again I get the Ble devices in scan list.

ic-twist, 2023-10-17:
@ReactRaylogic What is the issue, then, if you're able to get BLE devices in scan list?

ReactRaylogic, 2023-10-18:
The thing was after adding the permissions in Manifest I would have to go into the app setting and click on allow of Nearby Device permission (which was already added) to get the list.
I solved it by asking the permission of ACCESS_COARSE_LOCATION on runtime by the user.