innoveit/react-native-ble-manager #1022: Getting negotiated MTU on iOS

benedicksteve, 2023-06-06:
Would it be possible to get the negotiated MTU on iOS. Currently, the negotiated MTU can only be known on android after requesting the MTU, but as the iPhone does this automatically, there is currently no way (at least from what I have read) to know the MTU that has been negotiated between the iPhone and my peripheral.

In iOS, CBPeripherals has a function called [maximumWriteValueLength](https://developer.apple.com/documentation/corebluetooth/cbperipheral/1620312-maximumwritevaluelength) that can be used for this purpose.

marcosinigaglia, 2023-06-08:
Hi @benedicksteve , we added 2 functions in the latest release, we'll change the README soon.

closed 2023-06-08 by marcosinigaglia

stefnnn, 2023-10-09:
Hi @marcosinigaglia, awesome! can you point us to the new functions? I searched through the code, but didn't find out to read the current MTU for an active connection on iOS.

weisensee, 2024-10-16:
@marcosinigaglia we're also wondering what MTU has been negotiated, but I'm not seeing any functions to read MTU on iOS. Do you know if they exist?
Thanks!