innoveit/react-native-ble-manager #1025: Writing not working first try

ZenAl3x, 2023-06-09:
Hello there!

I've been having an issue that stops me from writing to a peripheral with my android device if the device is freshly restarted until I try 2 more times.

Steps to reproduce the behavior:
1. Connect to a peripheral and write something
2. Restart the android device and try connecting and writing again

Expected Behavior:
Connecting and writing first try

**Smartphone (please complete the following information):**
 - Device: Cat S42
 - OS: Android 12
 - react-native-ble-manager version: 10.0.2
 - react-native version: 0.70.5

I don't know if it's a bluetooth issue or something, but I really need help with this.

My code:

```javascript
    async connectAndPrint() {
          recursiveLimit++
        console.log('connect and print entrypoint')
        AsyncStorage.getItem('PRINTER')
          .then(async (storedPrinter) => {
            console.log("Retrieved printer", storedPrinter)
            if (storedPrinter == null) {
              console.error("NO PRINTER SAVED/ COULDN'T RETRIEVE FROM ASYNC STORAGE")
              return
            }

          console.log('Here is your printer before starting anything:', storedPrinter, "<-----")

          if(!storedPrinter) {
            console.log("No printer at all dude")
            return
          }
            const BMVService = "000018f0-0000-1000-8000-00805f9b34fb"
            const HPRTService = "0000ff00-0000-1000-8000-00805f9b34fb"
              const HPRTChars = "0000ff02-0000-1000-8000-00805f9b34fb"
              const BMVChars = "00002af1-0000-1000-8000-00805f9b34fb"
              var storedPrinterName = ""
              await AsyncStorage.getItem("PRINTER_NAME")
                  .then((res) => {
                      if (res) {
                          storedPrinterName = res
                      }
                  })
                  .catch((error) => {
                      throwError()
                      console.log(error);
                  });
              await BleManager.scan([], 1, true)
                  .then(() => console.log('Scan finished'))
                  .catch((err) => console.log(err))
              await BleManager.connect(storedPrinter)
                  .then(() => {
                      console.log("Connected");
                  })
                  .catch((error) => {
                      if (recursiveLimit >= 5) {
                          throwError()
                          recursiveLimit = 0
                      }
                      else this.connectAndPrint()
                      console.log(error);
                  });
              const buffer = Buffer.from(props.text);
              await BleManager.retrieveServices(storedPrinter)
                  .then((result) => {
                      console.log(result)
                  })
                  .catch((error) => {
                      if (recursiveLimit >= 5) {
                          throwError()
                          recursiveLimit = 0
                      }
                      else this.connectAndPrint()
                      console.log(error);
                  });
              await BleManager.write(
                  storedPrinter,
                  storedPrinterName.includes("BMV") ? BMVService : HPRTService,
                  storedPrinterName.includes("BMV") ? BMVChars : HPRTChars,
                  buffer.toJSON().data
              )
                  .then(() => {
                      props.loadingSetter(false)
                      props.loadingMessageTrigger(false)
                      console.log("Write: " + buffer.toJSON().data);
                      recursiveLimit = 0
                  })
                  .catch((error) => {
                      throwError()
                      console.log(error);
                  });
        })
        .catch((err) => console.log(err))
      }
```

marcosinigaglia, 2023-06-09:
I don’t think is the library. Check if you do something different. You have to scan the device before connect.

ZenAl3x, 2023-06-09:
> I don’t think is the library. Check if you do something different. You have to scan the device before connect.

I scan the device before connecting, check the code

```javascript
await BleManager.scan([], 1, true)
                  .then(() => console.log('Scan finished'))
                  .catch((err) => console.log(err))
```

closed 2023-12-04 by marcosinigaglia