innoveit/react-native-ble-manager #1027: Bonding stuck on "Only allow one bond request at a time" on Android

Koxx3, 2023-06-11:
**Describe the bug**
I didn't discover the exact conditions yet, but sometime, the bonding get stuck with this message :
"Only allow one bond request at a time"

It often happen when I "play" with multiple BLE devices and switch from one to another.

Considering the code : 
https://github.com/innoveit/react-native-ble-manager/blob/master/android/src/main/java/it/innove/BleManager.java#LL249C30-L249C67

It seems the bondRequest is not resetted in some conditions. Any idea ?

**To Reproduce**
Steps to reproduce the behavior:
1. Try to bond a device and kill the app ?

**Expected behavior**
Clear bond requests

**Smartphone (please complete the following information):**
 - Device: Pixel 6A, Xiaomi 5
 - OS: Android
 - react-native-ble-manager version: latests
 - react-native version: 0.70

Waryen, 2023-06-14:
I'm also having this issue, but the think is that the bonding doesn't have any issues but I have this message thrown in the console. Maybe I'm trying to bond multiple times to the same device, I'm going to investigate this.

marcosinigaglia, 2023-12-09:
This issue is stale because it has been open 180 days with no activity. Remove stale label or comment or this will be closed in 7 days.

Koxx3, 2023-12-10:
well ... the issue is still ongoing...

PR #1265 (2024-09-23): Feature/add docs notice
Adds further docs notice for Android behavior.

Fixes #1027 and #1246
closed 2024-09-23 by marcosinigaglia