innoveit/react-native-ble-manager #1034: Not receiving notifications and error on read() from Nordic UART Service peripheral

cahna, 2023-06-22:
This issue is related to a peripheral (a "smart" stopwatch) that is documented as using [Nordic UART Service (NUS)](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/libraries/bluetooth_services/services/nus.html) for bluetooth communication.

|Property|Value|
|---|---|
|Service UUID|6E400001-B5A3-F393-E0A9-E50E24DCCA9E|
|RX Characteristic|6E400002-B5A3-F393-E0A9-E50E24DCCA9E|
|TX Characteristic|6E400003-B5A3-F393-E0A9-E50E24DCCA9E|

**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
N/A

**Smartphone (please complete the following information):**
 - Device: [Fire HD 8 (10th generation)](https://developer.amazon.com/docs/fire-tablets/ft-device-specifications-firehd-models.html?v=firehd8_2020)
 - OS: Android 9, API Level 28 ([Fire OS 7.3.2.7](https://developer.amazon.com/docs/fire-tablets/ft-identifying-tablet-devices.html))
 - react-native-ble-manager version: `10.1.0`
 - react-native version: `0.71.8`

**Additional context**
- Hardware notes:
  - 2 peripherals have been used for testing: one used, one new, and both work with existing apps in the app stores (they are "smart" stopwatches)
- Potentially-related issues(?):
  -  Similar issues reading from device, but no details provided by the author: #1033
  - This could be similar, if my issue is related to the `write()` and/or `writeWithoutResponse()` calls (ie: not related to the notification listeners or `read()` call), but I am not sure, and there is no answer, yet: #1025
  - Worthy of note, but not directly related to my issue: I have noticed that the calls to `disconnect()` on android are not successful (no errors, but I cannot reconnect to the peripheral using another device until power-cycling the peripheral): #985

cahna, 2023-06-22:
Accidentally submitted before I completed writing the report... standby while I edit the original post. Thanks.

cahna, 2023-07-05:
Closing this. As -- sort of -- expected, my issue was not related to the library.

For reference: I found [a line of code in `Android-nRF-Toolbox`](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/05f6d9e680ca03968347d046f6d306bb015d31e9/profile_uart/src/main/java/no/nordicsemi/android/uart/data/UARTManager.kt#L110) where the MTU is explicitly set to `517`. After adding `await BleManager.requestMTU(pid, 517)`, everything started working.

closed 2023-07-05 by cahna

mrrogge, 2023-07-28:
Thank you so, so much for the extensive documentation on this issue! I've been battling a similar problem on an nRF5340 for a while now. In my case, the `requestMTU()` call would hang indefinitely, but once I added a 1sec delay after `connect()`, I started getting notifications.

I wonder if it would be worthwhile building an artificial delay right into the `connect()` method, to help alleviate these timing issues for others in the future. If I get a chance, I'll work on a PR.