innoveit/react-native-ble-manager #1035: BleManagerCentralManagerWillRestoreState not working as expected

gregg062, 2023-06-26:
I am trying to implement willRestoreState so that my app can wake up in the background by my peripherals event when it has been suspended, but I am unable to get BleManagerCentralManagerWillRestoreState to work, I have implemented a restoreIdentifierKey
```
 BleManager.start({
      showAlert: false,
      restoreIdentifierKey: 'TreeInTheBackground',
    })
```

and added a listener for events

```
bleManagerEmitter.addListener(
      'BleManagerCentralManagerWillRestoreState',
      async (data: any) => {
        console.log('BleManagerCentralManagerWillRestoreState', data)
          notifee.displayNotification({
            title: 'BLE WillRestoreState',
            body: JSON.stringify(data),
          })
      },
    )
```

But when my peripheral sends out an event after being disconnected BleManagerCentralManagerWillRestoreState is never called.

I am currently testing this by toggling on/off airplane mode, I have ensured bluetooth is not toggled off when I toggle on airplane mode. 

Has anyone been able to get willRestoreState to work?

MahmoudNafiseh, 2023-09-15:
Did you end up figuring this out? Would appreciate any help

gregg062, 2023-09-27:
@Oyveloper i have had that happen on occasion as well, unfortunately I haven't been able to find a solution....yet.

Oyveloper, 2023-10-04:
We have noticed that the callback is only called after you first call `BleManager.start` and `BleManager.checkState` again when the app wakes up. We have been able to reliably wake the app up and receive the willRestoreState callback, but we do not get any events on updated characteristics (we get them when in foreground)
When we start up in background now it seems that `retrieveServices` and `startNotification` just hang, and eventually the execution times out …

MahmoudNafiseh, 2023-10-05:
@marcosinigaglia Is it possible to look into this please?

Oyveloper, 2023-10-06:
We have managed to get a somewhat working solution. Our app now correctly wakes up when the connected device sends an update on a characteristic, or when it connects/disconnects with the device. The steps we found to work is the following:

1. When the app wakes, set up all listeners on the bleManagerEmitter. 
2. Call `BleManager.start()` and `BleManager.checkState()`. We observed that the `willRestoreState` callback is triggered as soon as we call `checkState`
3. Always call connect on the device, even if BleManager says it is connected. 
4. Do not call `retrieveServices` and `startNotification` if the device was not wakened due to a new connection. We observed that this is where our app started to hang and therefore timed out eventually. 

Further, we notice that when the app is wakened up because of a notification on a characteristic, our listener on value changes is not called for that event. We suspect that this is because our listener to the bleManagerEmitter is not able to be set up early enough, and the event is therefore lost there, even though it probably makes its way to the delegate method implemented by the native module. However, subsequent notifications are received.

Oyveloper, 2023-10-11:
We found a solution that worked for us, by simply adding the BleManager as the delegate for the peripheral during the `willRestoreState` callback. Added a fix in #1083

closed 2023-10-12 by marcosinigaglia

cheehieu, 2023-10-16:
@Oyveloper, nice work! Can you elaborate more on how you went about testing this? Does it relaunch the app in the background and trigger `willRestoreState` in all cases as described in the [developer docs](https://developer.apple.com/documentation/technotes/tn3115-bluetooth-state-restoration-app-relaunch-rules)?

I'm trying to get this working in my app, but it has been difficult to test. Did you restart your device to see `willRestoreState` triggered? Or wait (hours, days??) until iOS kills the app?

Oyveloper, 2023-10-19:
@cheehieu The easiest way of testing this is to enable `Fast App Terminiation` under developer settings on your iPhone. This makes it so that iOS sends your app into the `killed` state after ~10s, as opposed to ~8hours which is more typical. Note that there are two "killed" states in iOS. One is when the user actively removes your app using the app switcher. You will not receive events in this case. The other case is when iOS kills your app to save energy.

With this setup you can leave your app in the background for around 10s and then trigger some BLE event (disconnect, Characteristic event etc.) 
Your app will then be woken up, and the willRestoreState should be called (as long as you ensure that you subscribe to these events as soon as the app wakes)
It should also trigger any listeners for characteristic updates etc. 

Note: testing this while the app is connected to Metro has been very problematic for us. The issue seems that it then initiates a new bundle, and simply spends too much time. In order to test this better we made a setup where we log everything that happens to `AsyncStorage` so that we can test while disconnected from Metro, but still retrieve the log afterwards for troubleshooting.

vhakulinen, 2024-11-02:
Did any of you get the restore events to land on the javascript event listeners?

I noticed that the actual event wasn't emitted to the `RCTEventEmitter` but instead to `NSNotificationCenter`. You could receive these events from native code but not from js.

I made #1282 to fix this.

PR #1083 (2023-10-12): Set peripheral delegate when restoring state
I noticed when setting up handling events in the background that we did not receive the events for updateCharacteristics etc. After some digging it turns out that the delegate for the connected peripherals is not set, and as far as I have found, connecting when the react-native code is running seems to be too late to receive the characteristics update. 

Turns out that setting `peripheral.manager = self` in the `willRestoreState` handler solved the issue, and all events end up in our typescript handlers.

Should resolve #1035