innoveit/react-native-ble-manager #1044: BleManager could not be started. [TypeError: Cannot read property 'start' of null]

scytye, 2023-07-13:
**Describe the bug**
I downloaded the Github example to start my project, but it did not start successfully on the emulator. Error: Cannot read property 'start' of null. I tried to package it into an app and test it on a real phone, but the results were directly flashed back！！

**Expected behavior**

I hope to obtain the property  of  blemanager normally

**Screenshots**
This is my code
[image]

[image]

**Smartphone (please complete the following information):**
 - Device: [honor50SE]
 - OS: [android12]
 - react-native-ble-manager version: 10.1.2
 - react-native version:0.71.8

**Additional context**
I hope you can reply as soon as possible

closed 2023-07-13 by marcosinigaglia

scytye, 2023-07-14:
This is my package and library, it's not null !!!!!
[image]
[image]

scytye, 2023-07-14:
> If the library is null is not installed correctly.

Thanks for your reply. I tried reinstalling the library ,but the return value is still null. I printed the blemanager through console log, and the result was the constructor of  blemanager. Can I assume that I have installed it correctly? Here are my steps：
[image]
[image]

marcosinigaglia, 2023-07-14:
The js part is installed but probably the problem is the linking of the native part in you android project.

scytye, 2023-07-16:
>

scytye, 2023-07-16:
> The js part is installed but probably the problem is the linking of the native part in you android project.

hello sir
I checked my RN project and it was created by expo without the 'androidManifest' file. Can I modify it through the packaging tool? Or is there any way to monitor whether the native part is linked in my project