innoveit/react-native-ble-manager #1045: [Android] I can't run the app

carlosq-ordering, 2023-07-17:
**Describe the bug**
When I try to run `yarn android` an error appears in my console.

**Expected behavior**
Run successfully my react native application in Android.

**Screenshots**
1).
[image]

2).
[image]

**Smartphone (please complete the following information):**
 - Device: none
 - OS: Android
 - react-native-ble-manager version: ^10.1.2
 - react-native version: 0.64.4

**Additional context**
```JS
System:
    OS: macOS 13.4.1
    CPU: (10) arm64 Apple M1 Pro
    Memory: 79.89 MB / 16.00 GB
    Shell: 3.2.57 - /bin/bash
  Binaries:
    Node: 16.0.0 - ~/.nvm/versions/node/v16.0.0/bin/node
    Yarn: 1.22.19 - /usr/local/bin/yarn
    npm: 7.10.0 - ~/.nvm/versions/node/v16.0.0/bin/npm
    Watchman: Not Found
  Managers:
    CocoaPods: 1.11.3 - /usr/local/bin/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 22.4, iOS 16.4, macOS 13.3, tvOS 16.4, watchOS 9.4
    Android SDK: Not Found
  IDEs:
    Android Studio: 2022.2 AI-222.4459.24.2221.9971841
    Xcode: 14.3.1/14E300c - /usr/bin/xcodebuild
  Languages:
    Java: 11.0.14.1 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 17.0.1 => 17.0.1
    react-native: ^0.64.2 => 0.64.4
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found
```

draggie, 2023-07-20:
I also have similar setup but with RN 63 and the same error shows up - tried to migrate from older plx library to this one but no luck so far

marcosinigaglia, 2023-07-20:
You are missing the android SDK, is not a library problem, try to create a new empty project and go from there.

closed 2023-07-20 by marcosinigaglia

draggie, 2023-07-20:
Not really - this is related to older versions of project setup - what helps is to explicitly set the kotlin version dependency like in this thread: https://stackoverflow.com/questions/55460183/react-native-android-build-package-does-not-exist

Eozinum, 2023-08-08:
Someone found a solution? I faced also this issue trying to migrate from ble-plx

vinicius20m, 2023-10-18:
> Someone found a solution? I faced also this issue trying to migrate from ble-plx. With version 10.0.2 all works, but next versions cause such errors like on prev screens

Thanks, I downgraded to this version and worked

ekoooo, 2023-11-21:
same issues, `react-native-ble-manager@10.1.5` `react-native@0.62.3`