innoveit/react-native-ble-manager #1055: BleManagerDidUpdateValueForCharacteristic is not being triggered

Srikanth-Enuguru, 2023-08-01:
**Describe the bug**
I'm trying to listen for the changes happening on a characteristic. To achieve that I added a listener on BleManagerDidUpdateValueForCharacteristic event. But this handler is not being called when the characteristic value is changed.

**Code changes**
```
await BleManager.connect(peripheral.id);
await sleep(900);
await BleManager.retrieveServices(peripheral.id);

BleManager.startNotification(
          peripheral.id,
          "00001530-1212-efde-1523-785feabcd123",
          "00001531-1212-efde-1523-785feabcd123"
        )
          .then(async res => {
            console.log('started Notification');
            bleManagerEmitter.addListener(
              "BleManagerDidUpdateValueForCharacteristic",
              handleUpdateValueForCharacteristic,
            );
          })
          .catch((error) => {
            // Failure code
            console.log(error);
          });

BleManager.write(
            peripheral.id,
          "00001530-1212-efde-1523-785feabcd123",
          "00001531-1212-efde-1523-785feabcd123",
            buffer.toJSON().data
          )
            .then(async(data) => {
              // Success code
              console.log("Write: " + data);

              BleManager.readDescriptor(
                peripheral.id,
              "00001530-1212-efde-1523-785feabcd123",
              "00001531-1212-efde-1523-785feabcd123",
              "2902"
              )
                .then((data) => {
                  // Success code
                  console.log("Read after write 1: " + data);
                })
                .catch((error) => {
                  // Failure code
                  console.log(error);
                });
            })
            .catch((error) => {
              // Failure code
              console.log(error);
            });

const handleUpdateValueForCharacteristic = (
    data: BleManagerDidUpdateValueForCharacteristicEvent,
  ) => {
    console.debug(
      `[handleUpdateValueForCharacteristic] received data from '${data.peripheral}' with characteristic='${data.characteristic}' and value='${data.value}'`,
    );
  };
```

**Expected behavior**
BleManagerDidUpdateValueForCharacteristic should be triggered whenever the Characteristic value is changed.

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: Android 12
 - react-native-ble-manager version: 10.1.2
 - react-native version: 0.71.2

Please help me in fixing this issue.

Srikanth-Enuguru, 2023-08-01:
@marcosinigaglia Could you please help me.

inventivemedia, 2023-08-04:
I get the same error and not receiving the updated values from the device.
I am using Galaxy S21 with Android version 13 and API 33
 in my case what I get is
1) startNotification (BLE Manager)
2) registerNotify (BLE Manager)
3) setCharacteristicNotification() - uuid:####### enable: true (BluetoothGatt)
4) Characteristic ######## set NOTIFY (BLE Manager)
5) setCharacteristicNotification() - uuid: ########## enable: true (BluetoothGatt)
6) onDescriptorWrite success (BLE Manager)
7) Command queue empty (BLE Manager)

After all the above steps it should call "onCharacteristicChanged: " (BLE Manager)

@marcosinigaglia can you please tell me what could be the issue?

tonyle-geocomply, 2023-08-07:
I had the same issue. : (
Device: Android 12
react-native-ble-manager version: 10.1.2
react-native version: 0.71.2

BleManagerDidUpdateValueForCharacteristic did not trigger when startNotification

inventivemedia, 2023-08-08:
@marcosinigaglia what could be the issue? onCharacteristicChanged is not getting fired but write descriptor gives success

tonyle-geocomply, 2023-08-08:
On IOS device BleManagerDidUpdateValueForCharacteristic can listen the data, but on Android device, I had this problem for the above phone Galaxy s21

inventivemedia, 2023-08-10:
@marcosinigaglia is there a way we can debug and see what's the error? callback is not getting triggered

tonyle-geocomply, 2023-08-10:
@Srikanth-Enuguru, you still had issues with your Android device? Try another android version ?
Device: Android 12
react-native-ble-manager version: 10.1.2
react-native version: 0.71.2

marcosinigaglia, 2023-08-10:
Hi all, I test again on android 12/13 and is working fine. Check your permission and if your code is correct and if you can use a testing ble app to check if your device is working as expect.

inventivemedia, 2023-08-11:
@marcosinigaglia thanks for your message. so what I have noticed is: I have used your example file for connection, notification etc... 
1) it connects
2) get services
3) read descriptor values
4) start notification -> I get success but straight after I get BleManagerDisconnectPeripheral event (disconnected)

Thus data is not received in the app.

any idea? @tonyle-geocomply @Srikanth-Enuguru is that same happening with you guys?

inventivemedia, 2023-08-11:
as soon as starting the notification it seems that GATT client closes the connection.
"onConnectionStateChange" getting newState as 0 -> Disconnected

toanleviet95, 2023-08-13:
@inventivemedia  I resolved my problem. Just add requestMTU (Only on Android) before startNotification then it worked :D
```
BleManager.requestMTU(chosenPeripheral, 512) 
then 
BleManager.startNotification
```
Thank you @marcosinigaglia

inventivemedia, 2023-08-14:
@toanleviet95 I have tried but it still does not work, the maximum set MTU from the device I get is 150 bytes, do not think this would be an issue when getting data back from device... the problem is as soon as it starts notification mobile phone gets disconnected using bluetooth.

inventivemedia, 2023-08-19:
@marcosinigaglia anything you could suggest?

inventivemedia, 2023-08-24:
@marcosinigaglia checked with other phone, Google pixel 6 which has 5.2 bluetooth version, API level 33 and android 13. It does not work for it too! seems an issue with higher version, even changing MTU does not work..Can you please suggest? 
@Srikanth-Enuguru have you found any solution? 
@toanleviet95 which bluetooth version you have? and API level

arod7533, 2023-10-26:
I'm on Android 13 using API 33 with a Samsung A04e and have the same issue as above. Logcat shows this....

startNotification (BLE Manager)
registerNotify (BLE Manager)
setCharacteristicNotification() - uuid:####### enable: true (BluetoothGatt)
Characteristic ######## set NOTIFY (BLE Manager)
setCharacteristicNotification() - uuid: ########## enable: true (BluetoothGatt)
onDescriptorWrite success (BLE Manager)
Command queue empty (BLE Manager)

Everything seems to be setup correctly, but onCharacteristicChanged callback never gets invoked. 

@inventivemedia @Srikanth-Enuguru @toanleviet95  did you have any success in fixing this? 

@marcosinigaglia anything you can recommend? Much appreciated. I have tried overloading the onCharacteristicChanged method with the new version in API 33 with the byte[] value parameter, but still have not had much success. I'm happy to help test any potential fixes if you have any guidance.

tonyle-geocomply, 2023-12-06:
> BleManager.requestMTU(chosenPeripheral, 512) 
> then 
> BleManager.startNotification

My case was fixed by
```
BleManager.requestMTU(chosenPeripheral, 512) 
then 
BleManager.startNotification
```

closed 2024-03-13 by marcosinigaglia