innoveit/react-native-ble-manager #1056: Unexpected error using writeWithoutResponse on ios device, any kind soul can help?

Flano-123, 2023-08-02:
I am trying to use this package to manually send an audio to my AirPods, and there is no problem when it is operated like README at the beginning, but when I actually send data, the following abnormal error begins to appear.

**Describe the bug**
Sending actually data using writeWithoutResponse returns this error:

Error: Exception in HostFunction: Malformed calls from JS: field sizes are different.

[[86,114,84],[0,9,0],[["{\"type\":\"log\",\"level\":\"log\",\"mode\":\"BRIDGE\",\"data\":[\"data is array ? true\"]}",1],[1140,100,1690956656655,false]],8756]   
Error: Exception in HostFunction: Malformed calls from JS: field sizes are different.

[[86,114,84],[0,9,0],[["{\"type\":\"log\",\"level\":\"log\",\"mode\":\"BRIDGE\",\"data\":[\"data is array ? true\"]}",1],[1140,100,1690956656655,false]],8756]   
    at nativeFlushQueueImmediate (native)
    at enqueueNativeCall (http://192.168.101.7:19000/index.bundle//&platform=ios&dev=true&hot=false:2724:43)
    at nonPromiseMethodWrapper (http://192.168.101.7:19000/index.bundle//&platform=ios&dev=true&hot=false:3892:42)
    at createTimer (http://192.168.101.7:19000/index.bundle//&platform=ios&dev=true&hot=false:29859:38)
    at setTimeout (http://192.168.101.7:19000/index.bundle//&platform=ios&dev=true&hot=false:29712:18)
    at anonymous (/Users/distiller/react-native/sdks/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:469:30)
    at reject (/Users/distiller/react-native/sdks/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:185:17)
    at doResolve (/Users/distiller/react-native/sdks/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:227:13)
    at Promise (/Users/distiller/react-native/sdks/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:82:14)
    at writeWithoutResponse (http://192.168.101.7:19000/index.bundle//&platform=ios&dev=true&hot=false:234000:27)
    at anonymous (http://192.168.101.7:19000/index.bundle//&platform=ios&dev=true&hot=false:234655:62)
    at tryCallTwo (/Users/distiller/react-native/sdks/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:61:9)
    at doResolve (/Users/distiller/react-native/sdks/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:216:25)
    at Promise (/Users/distiller/react-native/sdks/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:82:14)
    at writeWithoutResponse (http://192.168.101.7:19000/index.bundle//&platform=ios&dev=true&hot=false:234654:27)
……

**To Reproduce**
Below is my code in JavaScript:

……
const base64String = await readFileAsStringAsync(fileUri, { encoding: 'base64' });
    
const data = Buffer.from(base64String, 'base64').toJSON().data;
   
await BleManager.writeWithoutResponse(peripheralId,serviceUUID,characteristicUUID,data)
……

**Expected behavior**
The data should sent successfully.

**Smartphone (please complete the following information):**
 - Device: iPhone 12
 - OS: 16.5.1
 - react-native-ble-manager version: ^10.1.2
 - react-native version: 0.71.8
 - react version: 18.2.0

**Additional context**
A similar problem was found in the previous issue and it said I should pass plain array instead of typed array like uint8array. But what I'm using is plain array the same to docs in my opinion. 

Can someone tell me or help me see what's wrong? Or am I just incorrect to want to send audio to my headphones and play it in this way? Will be appreciated a lot.

marcosinigaglia, 2023-12-08:
Can you paste a data example.

closed 2023-12-28 by marcosinigaglia

huyuyang, 2025-01-02:
@Flano-123 Did you fix this problem? I have the same problem as well