innoveit/react-native-ble-manager #1059: BleManagerDiscoverPeripheral undefined id

jonjow, 2023-08-09:
When listening on BleManagerDiscoverPeripheral I get the peripheral data but it does not contain an id, name or rssi however, it is showing the advertising payload.

```
    {
      advertising: {
        isConnectable: true,
        localName: 'KX2138014',
        manufacturerData: {
          CDVType: 'ArrayBuffer',
          bytes: [Array],
          data: 'AgEGBQMAGAoYEQcIufnXUKQgiXdAy/0swYBICwlLWDIxMzgwMTQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=',
        },
        serviceData: {},
        serviceUUIDs: [
          '1800',
          '180a',
        ],
        txPowerLevel: -2147483648,
      },
    }
```

**To Reproduce**
Steps to reproduce the behavior:
1. Subscript to the BleManagerDiscoverPeripheral
2. Scan for devices
3. The output is as above

**Smartphone (please complete the following information):**
 - Device: ASUS ROG PHONE 6
 - OS: Android 13
 - react-native-ble-manager version:  
 - react-native version:

```
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "B/9MABICAAMAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIAInYX+GG+VyIlV2kTPDKfBNBiqdeQziywAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": true, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "AgEaAgoICv9MABAFIZjD6WIAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": 8}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIAIWcHH4zBJJJpH+l9p+D5CUiHs6jpoe8wAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIALAeDOiUnUmuNKFaxrYPxrSdFpbmJ7sHwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIAIm1dvfjurpUmVlxoAGvIXQ4PdBUie8XQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIAJM/hilBTVAISB1iR1zzwWpNWFNPGoqCAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIALfQJchj7FVNzGU3D/3Us3lQXKiSHBBbAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "B/9MABICAAMAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIAInYX+GG+VyIlV2kTPDKfBNBiqdeQziywAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIAIWcHH4zBJJJpH+l9p+D5CUiHs6jpoe8wAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIALAeDOiUnUmuNKFaxrYPxrSdFpbmJ7sHwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  handleScanResults ID undefined
 LOG  {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIAIm1dvfjurpUmVlxoAGvIXQ4PdBUie8XQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}}
 LOG  deviceTimeout [{"id": undefined, "name": undefined, "timeout": 5}]
```

marcosinigaglia, 2023-08-09:
Hi @jonjow is fixed in the latest release.

jonjow, 2023-08-10:
@marcosinigaglia I am now using "react-native-ble-manager": "^10.1.3" and the BleManagerDiscoverPeripheral payload is now an empty object. Is there something else I need to setup or?

marcosinigaglia, 2023-08-10:
Check the log if you have error and if you have the connection permission.

jonjow, 2023-08-10:
@marcosinigaglia there is no error. The permissions are as before when working unless there is something new to be aware of?

jonjow, 2023-08-10:
I've added a check on the advert permission to an dis also fine:

```
  const requestBluetoothPermissions = async () => {
    try {
      const granted = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
        {
          title: 'Location Permission',
          message:
            'App needs access to your location to scan for Bluetooth devices.',
          buttonPositive: 'OK',
        },
      );

      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
        const advertise_granted = await PermissionsAndroid.request(
          PermissionsAndroid.PERMISSIONS.BLUETOOTH_ADVERTISE,
          {
            title: 'Advertise Permission',
            message: 'App needs access to your Advertise Permission.',
            buttonPositive: 'OK',
          },
        );
        if (advertise_granted === PermissionsAndroid.RESULTS.GRANTED) {
          console.log('Bluetooth permissions granted.');
          // You can start Bluetooth scanning here
          scanForDevices();
        } else {
          console.log('Bluetooth Advertise permissions denied.');
          // Handle the case when permissions are denied
        }
      } else {
        console.log('Bluetooth Location permissions denied.');
        // Handle the case when permissions are denied
      }
    } catch (error) {
      console.error('Error requesting Bluetooth permissions:', error);
    }
  };
```

marcosinigaglia, 2023-08-10:
You need BLUETOOTH_CONNECT

jonjow, 2023-08-10:
> BLUETOOTH_CONNECT

    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation"/>
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />

I have all these.

Added another check and it's all passed:

```
  const requestBluetoothPermissions = async () => {
    try {
      const granted = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
        {
          title: 'Location Permission',
          message:
            'App needs access to your location to scan for Bluetooth devices.',
          buttonPositive: 'OK',
        },
      );

      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
        const connect_granted = await PermissionsAndroid.request(
          PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
          {
            title: 'Connect Permission',
            message: 'App needs access to your Connect Permission.',
            buttonPositive: 'OK',
          },
        );
        if (connect_granted === PermissionsAndroid.RESULTS.GRANTED) {
          const advertise_granted = await PermissionsAndroid.request(
            PermissionsAndroid.PERMISSIONS.BLUETOOTH_ADVERTISE,
            {
              title: 'Advertise Permission',
              message: 'App needs access to your Advertise Permission.',
              buttonPositive: 'OK',
            },
          );
          if (advertise_granted === PermissionsAndroid.RESULTS.GRANTED) {
            console.log('Bluetooth permissions granted.');
            // You can start Bluetooth scanning here
            scanForDevices();
          } else {
            console.log('Bluetooth Advertise permissions denied.');
            // Handle the case when permissions are denied
          }
        } else {
          console.log('Bluetooth Connect permissions denied.');
          // Handle the case when permissions are denied
        }
      } else {
        console.log('Bluetooth Location permissions denied.');
        // Handle the case when permissions are denied
      }
    } catch (error) {
      console.error('Error requesting Bluetooth permissions:', error);
    }
  };
```

jonjow, 2023-08-10:
@marcosinigaglia i've rolled back to "react-native-ble-manager": "10.0.2" and this works so i'll continue with this until it's fixed.