innoveit/react-native-ble-manager #1061: calling read() to get the device serial for pin pairing sometimes doesn't prompt the pin modal

cfradella, 2023-08-17:
**Describe the bug**
On iOS, calling `read()` to get the serial characteristic from the device, normally prompting the pin pairing modal, occasionally returns `null` as though the modal came up and was cancelled/wrong pin typed in. It fails about 25% of the time, and succeeds 75% of the time.

**To Reproduce**
Steps to reproduce the behavior:
Calling the following line: `const char = await BleManager.read(deviceId, serviceUUID, characteristicUUID);`

**Expected behavior**
Calling this `read()` prompts the user to type in a pairing pin

**Smartphone (please complete the following information):**
 - Device: iPhone 13 mini
 - OS: iOS 16.6
 - react-native-ble-manager version: 10.1.3
 - react-native version: 0.72.3

**Additional context**
We've made sure `retrieveServices()` is always called before each `read()` call.

marcosinigaglia, 2024-03-04:
This issue is stale because it has been open 180 days with no activity. Remove stale label or comment or this will be closed in 7 days.

closed 2024-03-12 by marcosinigaglia

vhakulinen, 2024-03-15:
I'm witnessing similar behavior.

If the peripheral requires pairing before reading something but isn't accepting any pairing requests, the `read` resolves with `null` value.