innoveit/react-native-ble-manager #1064: startNotificationUseBuffer does not trigger event

eleroy, 2023-08-25:
Hello everyone,

I am trying to read notification at a rate of about 20 Hz. Unfortunately, the app does not run fast enough so I am trying to use the buffered notification.

My characteristic emits exacty 5 bytes every 50 ms (or so). When I use startNotificationUseBuffer with a buffer length of 1 or 5 it works well, but when I try 10, 15 or any larger number than 5, it does not work anymore (the event is not emitted).

I am using the library in an expo project with a development build.

react-native 0.71.7
"react-native-ble-manager": "^10.1.3"

eleroy, 2023-08-25:
Fix proposal in pull request #1065

closed 2023-08-25 by eleroy

gabrielfast, 2023-09-07:
Hi guys,

From the documentation, I understand that the logic should be this (you can run the code on the website
https://www.jdoodle.com/online-java-compiler/)

```java
import java.nio.ByteBuffer;

public class MyClass {
  public static void main(String args[]) {
     ByteBuffer items = ByteBuffer.allocate(5);
     
     byte[] init = {0};
     byte[] value = {1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11};
     byte[] toInsert = null;
     byte[] rest = null;

     // initial values
     items.put(init);
      
     if (value.length > items.remaining()) {
        int restLength = value.length - items.remaining();
        rest = new byte[restLength];
        toInsert = new byte[items.remaining()];
        System.arraycopy(value, 0, toInsert, 0, toInsert.length);
        System.arraycopy(value, toInsert.length, rest, 0, rest.length);
     } else {
        toInsert = value;
     }
     
        System.out.println("toInsert");
        for (int i = 0; i < toInsert.length; i++) {
            System.out.println(i + ": " + toInsert[i]);
        }

        System.out.println("rest");
        for (int i = 0; i < rest.length; i++) {
            System.out.println(i + ": " + rest[i]);
        }
   }
}
```

The array has size 5 and I already have 1 so it should only accept 4 more values to complete the buffer and the other values are returned.

```
toInsert
0: 1
1: 2
2: 3
3: 4
rest
0: 5
1: 6
2: 7
3: 8
4: 9
5: 10
6: 11
```

Here is the fix already tested on Android:

```java
package it.innove;

import java.nio.ByteBuffer;

public class NotifyBufferContainer {
    public ByteBuffer items;

    public NotifyBufferContainer(int size) {
        this.items = ByteBuffer.allocate(size);
    }
    public void resetBuffer(){
        this.items.clear();
    }
    public byte[] put(byte[] value){
        byte[] toInsert = null;
        byte[] rest = null;
        
        if (value.length > this.items.remaining()) {
            int restLength = value.length - this.items.remaining();
            rest = new byte[restLength];
            toInsert = new byte[this.items.remaining()];
            System.arraycopy(value, 0, toInsert, 0, toInsert.length);
            System.arraycopy(value, toInsert.length, rest, 0, rest.length);
        } else {
            toInsert = value;
        }
        
        this.items.put(toInsert);

        return rest;
    }
    public boolean isBufferFull(){
        return this.items.remaining() == 0;
    }
    public int size() {
        return this.items.limit();
    }
    @Override 
    protected void finalize() throws Throwable {
        this.items = ByteBuffer.allocate(0);
    }
}
```