innoveit/react-native-ble-manager #1066: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@be0b165f:GattService resisterScanner

GokulKir, 2023-08-25:
**Before open an issue**
- Check the closed issues, your question maybe is not new.
- We can't debug or reproduce hardware issue.
- If the library is not working 99% you did something wrong, in the code, installation or in the phone permissions.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - react-native-ble-manager version: 
 - react-native version:

**Additional context**
Add any other context about the problem here.

lucasmrau, 2023-09-06:
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
  package="com.package name"> // replace package name with yours

/* add the permissions below before the aplication tag */
  <uses-permission android:name="android.permission.INTERNET" />
  <uses-permission android:name="android.permission.BLUETOOTH" />
  <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
  <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />

lucasmrau, 2023-09-06:
or, in android 13:

<uses-permission android:name="android.permission.BLUETOOTH" 
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" 
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" 
<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" 
<uses-permission android:name="android.print.BLUETOOTH" 
<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" 
<uses-permission android:name="android.permission.BLUETOOTH_DEBUG" 
<uses-feature android:name="android.hardware.bluetooth" android:required="true"

AmilaJJ, 2023-09-08:
Request BLUETOOTH_SCAN permissions on run time from the user

ReactRaylogic, 2023-10-17:
I am asking user for BLUETOOTH_SCAN permission and location permission on run time but until and unless i go into permission and click on Allow of already allowed permission of Near by devices the ble device does not show in listner of ble scan.

closed 2023-12-04 by marcosinigaglia