innoveit/react-native-ble-manager #1071: retrieveServices() sometimes returns empty array of characteristics and services

cfradella, 2023-09-06:
Calling `retrieveServices()` before attempting to pair with a device will occasionally return an empty array for both the `characteristics` and `services` properties. If I wait ~3 seconds and call again, that same response comes back except with the expected characteristics and services within the array. This specifically happens after a user fails the pin pairing process.

Calling `removePeripheral` or `removeBond` dont seem to have any affect on this. I believe this is only happening for Android at the moment.

Example response with no characteristics:

```
{
    "advertising": {
        "isConnectable": true,
        "localName": "Device",
        "manufacturerData": {
            "CDVType": "ArrayBuffer",
            "bytes": [Array],
            "data": "AdEGBglFbnNvKxEHfz94BKQSQX29+9TnATTlAQAAAAAAAA="
        },
        "serviceData": {},
        "serviceUUIDs": ["02e53401-e7d4-fbbd-7d41-12a404383f7f"],
        "txPowerLevel": -2147483648
    },
    "characteristics": [],     <====
    "id": "F1:0A:EF:BB:7C:11",
    "name": "Device",
    "rssi": -66,
    "services": []     <====
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to pair to a device with a pin
2. Fail the pain pair
3. Immediately call `retrieveServices()`
4. It returns no characteristics

**Expected behavior**
Characteristics and services should be retrieved 100% of the time.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: Pixel 6a
 - OS: Android 13
 - react-native-ble-manager version: 10.1.3
 - react-native version: 0.72.4

marcosinigaglia, 2023-12-11:
Hi @cfradella ,  why do you think is a library issue?

Loovery, 2024-02-08:
@marcosinigaglia Because this never happened to me when the version was written in ObjectC (8.x.x). Now I face this problem with intermittent success :(

closed 2024-05-17 by marcosinigaglia