innoveit/react-native-ble-manager #1074: TypeError: Cannot read property 'start' of null

Johno95CZ, 2023-09-18:
**Describe the bug**
Every time I call BleManager.start() I get an error "TypeError: Cannot read property 'start' of null". I tried to reinstall the whole library, attempted to create a new expo project, and installed only this library, etc. But always the same error. I even tried every tips in already created issues on this GitHub repo.

**Screenshots**
[image]

[image]

**Smartphone (please complete the following information):**
 - Device: Samsung galaxy A72
 - OS: Android 13
 - react-native-ble-manager version: ^10.1.3
 - react-native version: 0.72.4
 - expo: ^49.0.11

closed 2023-09-19 by marcosinigaglia

Johno95CZ, 2023-09-19:
I'm sorry @marcosinigaglia but can you show me an open issue that addresses this or an issue where it shows how to fix this?

marcosinigaglia, 2023-09-19:
@Johno95CZ  there are tons of similar issues, the library is not installed correctly.

Johno95CZ, 2023-09-19:
Then what is the correct way? I feel like I tried all the ways already. Even read all similar issues. I even tried to install it manually.

marcosinigaglia, 2023-09-19:
Read the expo docs, is not related to this library.

kir1m1, 2024-05-06:
I was able to solve this by substitutin the file uri for the `react-native-ble-manager` dev dependency within the `example/package.json` with it's current npm version.