innoveit/react-native-ble-manager #1075: BleManager.write() doesn't works on iOS whereas writeWithoutResponse() works.

ReactRaylogic, 2023-09-21:
**Describe the bug**
i am trying to use BLE PREPARE_WRITE  and EXECUTE_WRITE  to write a bulk data more than the default MTU using write() methods on iOS 16, where as writeWithoutResponse works but has a character limit of 20 it sends multiple string of size 20.

const write = (peripheral, string) => {
    return new Promise((resolve, reject) => {
      const buffer = Buffer.from(string);
      
      BleManager.write(
        peripheral.id,
       serviceUUID,
        charUUID,
        // encode & extract raw `number[]`.
        // Each number should be in the 0-255 range as it is converted from a valid byte.
        buffer.toJSON().data,
        string.length,
      ).then(resp => {
          resolve(1);
        })
        .catch(e => {
          console.log('send error', e);
          reject(e);
        });
    });
  };

getting Error as
**Writing is not permitted** 

**Expected behavior**
Should write to the Ble device.

**Smartphone (please complete the following information):**
 - Device: iPad
 - OS:  iOS 16
 - react-native-ble-manager version:  "^10.1.2"
 - react-native version:"0.71.11"

marcosinigaglia, 2023-09-27:
Is a ble device problem/configuration.

closed 2023-09-27 by marcosinigaglia

ReactRaylogic, 2023-09-29:
I am able to set it via an Android device MTU upto 512 can be set but the same routine on iOS it fails to set the MTU to required value.