innoveit/react-native-ble-manager #1095: Is writeValue required for iOS customized ble devices

Zoulindong999, 2023-11-07:
**Before open an issue**
- Check the closed issues, your question maybe is not new.
- We can't debug or reproduce hardware issue.
- If the library is not working 99% you did something wrong, in the code, installation or in the phone permissions.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - react-native-ble-manager version: 
 - react-native version:

**Additional context**
Add any other context about the problem here.

Zoulindong999, 2023-11-07:
https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518747-writevalue

marcosinigaglia, 2023-11-07:
Hi @Zoulindong999 , I don't understand your issue.

closed 2023-12-01 by marcosinigaglia