innoveit/react-native-ble-manager #1111: Eddystone beacons are not visible on Android 13

maksimLobachevskiy, 2023-12-05:
Scanning does not find any Eddystone beacons, all other bluetooth devices are visible. I used the sample code to test it.

**Smartphone (please complete the following information):**
 - Device: Samsung XCover 5
 - OS: Android 13
 - react-native-ble-manager version:  11.0.0
 - react-native version: 0.73.13

marcosinigaglia, 2023-12-05:
Hi @maksimLobachevskiy , check your manifest permission.

maksimLobachevskiy, 2023-12-05:
@marcosinigaglia thank you for quick response
this is my Androidmanifest
` <uses-permission android:name="android.permission.INTERNET"/>
    <uses-permission android:name="android.hardware.camera.any"/>
    <uses-permission android:name="android.permission.CAMERA"/>
    <uses-feature android:name="android.hardware.camera"/>
    <uses-permission android:name="android.permission.FLASHLIGHT"/>
    <uses-feature android:name="android.hardware.camera.flash" android:required="false"/>
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE"/>
    <uses-permission android:name="android.permission.VIBRATE"/>
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="30"/>`

marcosinigaglia, 2023-12-05:
I think the `neverForLocation` flag is filtering the beacon.

maksimLobachevskiy, 2023-12-05:
@marcosinigaglia unfortunately without the flag the scanner does not see any devices at all.

marcosinigaglia, 2023-12-05:
Do you see the beacons using the lighblue app?

maksimLobachevskiy, 2023-12-06:
Yes I can see beacons using apps

marcosinigaglia, 2023-12-06:
Can you try with older version?

closed 2023-12-06 by maksimLobachevskiy

marcosinigaglia, 2023-12-06:
@maksimLobachevskiy can you put your manifest in the issue so it remains a trace for those who have the same problem?

deCodeIt, 2023-12-17:
In my case when I wasn't able to scan any devices, "Turning on GPS" on my Android fixed the issue and I was able to see the devices