innoveit/react-native-ble-manager #1127: Background request sending but not getting anything back through listener

anton25360, 2023-12-07:
**Describe the bug**

Normal foreground behavior:
Send a `writeWithoutResponse` every 5s, get expected register back via the listener (`bleManagerEmitter.addListener("BleManagerDidUpdateValueForCharacteristic")`)

Background behavior (the issue):
Send a `writeWithoutResponse` every 5s when the app is in the bavkground, using [react-native-background-actions](https://github.com/Rapsssito/react-native-background-actions) library (the request does get sent) but the listener doesn't recieve anything back - is it the fault of the phone or the Bluetooth device?

**To Reproduce**
Code to reproduce the behavior:
```
  const BLEBackgroundTask = async () => {
    // Poll your Bluetooth device using react-native-ble-manager
    // Implement your Bluetooth polling logic here

    bleManagerEmitter.addListener(
      "BleManagerDidUpdateValueForCharacteristic",
      ({ value }) => {
        console.log("BG value", value);
      }
    );

    setInterval(() => {
      console.log("run BG task");

      getData("deviceBluetoothID").then(async (storedID) => {
        try {
          // Connect to the Bluetooth device
          await BleManager.connect(storedID);

          // Perform your Bluetooth operations (read, write, etc.)
          console.log("connected!");
          getDeviceInfo(); //writeWithoutResponse here

          await BleManager.disconnect(storedID);
        } catch (error) {
          console.error("Bluetooth error:", error);
        }
      });
    }, 5000);
  };

  // Configure background actions
  const options = {
    taskName: "BLEBackgroundTask",
    taskTitle: "Polling Bluetooth Device",
    taskDesc: "Polling Bluetooth device every 5 seconds",
    taskIcon: {
      name: "ic_launcher",
      type: "mipmap",
    },
    color: "#ff00ff",
  };

  // Start the background task
  BackgroundFetch.start(BLEBackgroundTask, options);
  useEffect(() => BackgroundFetch.stop(), []);
```

**Expected behavior**
I expect the listener to return the request response from the device

**Smartphone (please complete the following information):**
 - Device: iPhone 12
 - OS: iOS 17
 - react-native-ble-manager version: ^8.7.0
 - react-native version: 0.70.8

marcosinigaglia, 2023-12-07:
If is working in foreground is not a library problem.

closed 2023-12-07 by marcosinigaglia

luhaimin, 2023-12-13:
I also encountered the same problem
IOS: 15.8
React native: 0.67
React native le manager: 10+
It can be confirmed that the data has been sent to the device, as changes in the device's LED can be seen, but there is no response from the listening 'BleManagerDidUpdateValue ForCharacteristic'

luhaimin, 2023-12-14:
Very good. It has been confirmed that it has nothing to do with this library and is caused by the iOS system.