innoveit/react-native-ble-manager #1128: Android crash recieving data

paulcockrell, 2023-12-07:
**Before open an issue**
- Check the closed issues, your question maybe is not new.
- We can't debug or reproduce hardware issue.
- If the library is not working 99% you did something wrong, in the code, installation or in the phone permissions.

**Describe the bug**
Using the example app I added the following to the `connectPeripheral` function after the `BleManager.retrieveServices` 

```
if (peripheral.id === 'XX:XX:XX:XX:XX:XX') {
  console.log(`[connectPeripheral][${peripheral.id}][${peripheral.name}] Starting notifications.`);
  await BleManager.startNotification(peripheral.id, DEVICE_UUID, DEVICE_CHARACTERISTIC);
}
```

Once the device successfully is found and connected to, I then make it transmit data and the app will receive payload and crash

The edited output of `adb logcat` 
```
12-07 17:46:25.647 13597 13748 E AndroidRuntime: FATAL EXCEPTION: binder:13597_5
12-07 17:46:25.647 13597 13748 E AndroidRuntime: Process: it.innove.example.ble, PID: 13597
12-07 17:46:25.647 13597 13748 E AndroidRuntime: java.lang.StackOverflowError: stack size 991KB
12-07 17:46:25.647 13597 13748 E AndroidRuntime:        at it.innove.Peripheral.onCharacteristicChanged(Peripheral.java:423)
12-07 17:46:25.647 13597 13748 E AndroidRuntime:        at android.bluetooth.BluetoothGattCallback.onCharacteristicChanged(BluetoothGattCallback.java:159)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Clone example app
2. Install react-native-ble-manager in to packages, and install packages
3. Add the code snippet from above, into the connectPeripheral function in App.tsx, after the call to `BleManager.retrieveServices`
5. Save and run app with `yarn android`
6. Search and connect to device that transmits notifications 
7. Make device transmit payload 
8. Watch App crash

**Expected behavior**
The app does not crash from a stackoverflow exception when receiving data.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: Pixel 7a
 - OS: 14
 - react-native-ble-manager version: 11.0.4 
 - react-native version: 0.72.6

**Additional context**
I have tried this with mutliple BLE devices. Both send small packets of data. Will always crash.

marcosinigaglia, 2023-12-07:
Hi @paulcockrell can you try a 10.x.x version? Did you try with other android version os?

marcosinigaglia, 2023-12-08:
It the version 11 working with the jdk 11?

paulcockrell, 2023-12-08:
Unfortunately version 11 crashes when using JDK 11

marcosinigaglia, 2023-12-08:
What is your targetSdk?

paulcockrell, 2023-12-08:
In the adroid/build.gradle file it has the following, with targetSdk set to 33. Is this what you were after?
```
android {
        compileSdk safeExtGet("compileSdk", 33)

        compileOptions {
                sourceCompatibility JavaVersion.VERSION_1_9
                targetCompatibility JavaVersion.VERSION_1_9
        }

        defaultConfig {
                minSdkVersion safeExtGet("minSdkVersion", 21)
                targetSdk safeExtGet("targetSdk", 33)
        }
        lintOptions {
                abortOnError false
        }
}
```

PrestonFocusV, 2023-12-09:
Also crashing for us on version 11 with Expo SDK 49:

```
 ERROR  Your app just crashed. See the error below.
java.lang.StackOverflowError: stack size 991KB
  it.innove.Peripheral.onCharacteristicRead(Peripheral.java:472)
  android.bluetooth.BluetoothGattCallback.onCharacteristicRead(BluetoothGattCallback.java:108)
  it.innove.Peripheral.onCharacteristicRead(Peripheral.java:479)
  it.innove.Peripheral.onCharacteristicRead(Peripheral.java:472)
  android.bluetooth.BluetoothGattCallback.onCharacteristicRead(BluetoothGattCallback.java:108)
```