innoveit/react-native-ble-manager #1132: `startNotification` fails for 128-bit UUIDs on iOS

thehale, 2023-12-09:
**Before open an issue**
- [x] Check the closed issues, your question maybe is not new.

> The following old issues reflect similar usage problems. Many of them were closed en masse a few days ago without a clear resolution. The others recommend the workaround of "just use the short form of the UUID on iOS".
>   - https://github.com/innoveit/react-native-ble-manager/issues/716
>   - https://github.com/innoveit/react-native-ble-manager/issues/951#issuecomment-1420416828
>   - https://github.com/innoveit/react-native-ble-manager/issues/913
>   - https://github.com/innoveit/react-native-ble-manager/issues/793
>   - https://github.com/innoveit/react-native-ble-manager/issues/706
>   - https://github.com/innoveit/react-native-ble-manager/issues/394
>   - https://github.com/innoveit/react-native-ble-manager/issues/265
>   - https://github.com/innoveit/react-native-ble-manager/issues/259
>
> I probably missed several, but I think this list is impressive enough to show that the current behavior is unexpected/unintuitive.

- [x] We can't debug or reproduce hardware issue.
- [x] If the library is not working 99% you did something wrong, in the code, installation or in the phone permissions.

**Describe the bug**
Tracking notifications on iOS with full 128-bit UUIDs results in the error `Could not find service with UUID`. Testing the same on Android enables notifications successfully.

When using 16-bit UUIDs, both platforms work as expected.

**To Reproduce**
Steps to reproduce the behavior:

[128-bit UUIDs]

1. Call `BleManager.startNotification` on a Bluetooth Device which uses the Bluetooth Base UUID (`0000xxxx-0000-1000-8000-00805F9B34FB`) for one or more of its services/characteristics. Use the full 128-bit UUID
    ```typescript
    BleManager.startNotification(
        "189fd3cb-fb2b-a6a6-b214-246d82f81298",
        "0000aadb-0000-1000-8000-00805f9b34fb",
        "0000aadc-0000-1000-8000-00805f9b34fb"
    )
    ```
2. Platform Dependent Behavior
   - [Android] Observe that the notifications work as expected.
   - [iOS] Be surprised by the error `Could not find service with UUID 0000aadb-0000-1000-8000-00805f9b34fb on peripheral with UUID 189fd3cb-fb2b-a6a6-b214-246d82f81298`

[16-bit UUIDs]
1. Call `BleManager.startNotification` on a Bluetooth Device which uses the Bluetooth Base UUID (`0000xxxx-0000-1000-8000-00805F9B34FB`) for one or more of its services/characteristics. Use the shorthand 16-bit UUIDs
    ```typescript
    BleManager.startNotification(
        "189fd3cb-fb2b-a6a6-b214-246d82f81298",
        "aadb",
        "aadc"
    )
    ```
2. Platform Dependent Behavior
   - [Android] Observe that notifications work as expected.
   - [iOS] Observe that notifications work as expected.

> Note:
>
> If the service/characteristic UUIDs do not use the Bluetooth Base UUID, then they work just fine on iOS in their full 128-bit form.

**Expected behavior**
I expect both platforms to demonstrate the same behavior for both 128-bit UUIDs AND 16/32-bit shorthand UUIDs.

**Screenshots**
If applicable, add screenshots to help explain your problem.

n/a

**Smartphone (please complete the following information):**
 - Device: iPhone 6s, Google Pixel 4a
 - OS: iOS 15.8, Android 13
 - react-native-ble-manager version: 10.1.5
 - react-native version: 0.71.12

**Additional context**
Here's some of the relevant code I found:

Based on [`BleManager.swift`](https://github.com/innoveit/react-native-ble-manager/blob/f46dd5c303ea0e70a44e299fb1ebabff86da9edd/ios/BleManager.swift#L381-L385), it appears that the iOS version of `retrieveServices` creates a [`CBUUID` (Core Bluetooth UUID)](https://developer.apple.com/documentation/corebluetooth/cbuuid) which compresses 128-bit UUIDs to their 16/32-bit forms as specified in the [Bluetooth 4.0 specification, Volume 3, Part F, Section 3.2.1](https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=456433)

On Android, the [`UUIDHelper.java`](https://github.com/innoveit/react-native-ble-manager/blob/f46dd5c303ea0e70a44e299fb1ebabff86da9edd/android/src/main/java/it/innove/UUIDHelper.java) appears to be responsible for UUID conversions.

**Current workaround**

I can successfully connect to my Bluetooth devices by compressing my known 128-bit UUIDs on iOS to their shorter 16-bit versions prior to calling `startNotification`. 

> **Note:**
>
> This code only handles 16-bit UUIDs. Based on the docs for `CBUUID` (see *Additional context*), it looks like a full solution would also need to support 32-bit UUIDs

```javascript
function compressUUID(uuid: string): string {
  // Bluetooth 4.0 specification, Volume 3, Part F, Section 3.2.1
  // https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=456433

  // Required since react-native-ble-manager needs short UUIDs for notifications on
  // iOS, so I need to be able to compare them to the full UUIDs known for my devices.
  
  // CBUUID (iOS):
  // https://developer.apple.com/documentation/corebluetooth/cbuuid
  // Source: https://github.com/innoveit/react-native-ble-manager/blob/f46dd5c303ea0e70a44e299fb1ebabff86da9edd/ios/BleManager.swift#L381-L385
  const baseUUID = /0000(....)-0000-1000-8000-00805F9B34FB/;
  if (baseUUID.test(uuid.toUpperCase())) {
    return uuid.substring(4, 8);
  } else {
    return uuid;
  }
}

BleManager.startNotification(
    "189fd3cb-fb2b-a6a6-b214-246d82f81298",
    compressUUID("0000aadb-0000-1000-8000-00805f9b34fb"),
    compressUUID("0000aadc-0000-1000-8000-00805f9b34fb"),
)
```

swunder, 2023-12-13:
are there any solutions? I ahve the same problems with BleManager.read. See my question in a forum. 
[https://stackoverflow.com/questions/77648401/react-native-blemanager-startnotification-is-causing-a-crash-all-the-time](url)

thehale, 2023-12-13:
Yes @swunder. See the **workaround** cited in my question.

closed 2023-12-21 by marcosinigaglia

phongbksneep, 2024-03-01:
> Yes @swunder. See the **workaround** cited in my question.

How do you generate codes like this: /0000(....)-0000-1000-8000-00805F9B34FB/, I generated this format on this site, right? https://www .uuidgenerator.net/