innoveit/react-native-ble-manager #1141: 🐛 BLE Scanner Device not found on Android 13

C0zmaCatalin, 2023-12-16:
**Describe the bug**
I used the example code from library to test a BLE Scanner Device ( NETUM R2L, 2D Ring Barcode Scanner).
The problem it's that the device it's found on Android 9, but on Android 13 it's not there. I tested with other app from google play https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner&hl=en&gl=US, and the device it's found there.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
I just opened the app on both platforms, and the device it's not found on android 13

**Smartphone (please complete the following information):**
-Samsung Galaxy Note 8 (Android 9), 
-Google Pixel 6 (Android 13), 
-Netum R2L 2D Ring Barcode Scanner, iBeacon
 - react-native-ble-manager version:  11.0.6
 - react-native version: 0.73.0

**Additional context**
Environment info
info Fetching system and libraries information...
System:
  OS: macOS 13.5
  CPU: (8) arm64 Apple M2
  Memory: 88.30 MB / 8.00 GB
  Shell:
    version: "5.9"
    path: /bin/zsh
Binaries:
  Node:
    version: 20.10.0
    path: /usr/local/bin/node
  Yarn:
    version: 1.22.19
    path: /usr/local/bin/yarn
  npm:
    version: 10.2.3
    path: /usr/local/bin/npm
  Watchman: Not Found
Managers:
  CocoaPods:
    version: 1.14.3
    path: /usr/local/bin/pod
SDKs:
  iOS SDK:
    Platforms:
      - DriverKit 22.4
      - iOS 16.4
      - macOS 13.3
      - tvOS 16.4
      - watchOS 9.4
  Android SDK: Not Found
IDEs:
  Android Studio: 2023.1 AI-231.9392.1.2311.11076708
  Xcode:
    version: 14.3.1/14E300c
    path: /usr/bin/xcodebuild
Languages:
  Java:
    version: 17.0.9
    path: /usr/bin/javac
  Ruby:
[... 49 lines cut ...]
                                                                                                                  164,
                                                                                                                  226,
                                                                                                                  79,
                                                                                                                  177,
                                                                                                                  175,
                                                                                                                  207,
                                                                                                                  198,
                                                                                                                  235,
                                                                                                                  7,
                                                                                                                  100,
                                                                                                                  120,
                                                                                                                  37,
                                                                                                                  39,
                                                                                                                  81,
                                                                                                                  101,
                                                                                                                  193,
                                                                                                                  198 ],
                                                                                                               data: 'AhX9pQaTpOJPsa/PxusHZHglJ1FlwcY=',
                                                                                                               CDVType: 'ArrayBuffer' } },
                                                                                                         txPowerLevel: -2147483648,
                                                                                                         isConnectable: true,
                                                                                                         serviceData: {},
                                                                                                         localName: 'RT barcode scanner',
                                                                                                         serviceUUIDs: [ 'fff0' ],
```

marcosinigaglia, 2023-12-17:
It's a permission problem.

closed 2023-12-17 by marcosinigaglia

C0zmaCatalin, 2023-12-17:
What permission? 
like I said, it’s the code copy pasted from this GitHub repo! 
Tell me please what I need to update in the code because I didn’t modify anything! An also I approved all the permissions when I entered in the app.

maksimLobachevskiy, 2023-12-22:
@C0zmaCatalin I had similar problem with permissions on Android, first of all check your Android Manifest, should be (worked for me)
```
 <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
    <uses-permission android:name="android.permission.BLUETOOTH" tools:remove="android:maxSdkVersion"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" tools:remove="android:maxSdkVersion"/>
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28"/>
```

SyntaxError843, 2024-01-18:
For those who are still stuck try editing the manifest

`<manifest xmlns:android="http://schemas.android.com/apk/res/android" xmlns:tools="http://schemas.android.com/tools">`
`<uses-permission android:name="android.permission.BLUETOOTH_SCAN" tools:remove="android:usesPermissionFlags"/>`

krunalshoppiko, 2024-05-02:
> What permission? like I said, it’s the code copy pasted from this GitHub repo! Tell me please what I need to update in the code because I didn’t modify anything! And also I approved all the permissions when I entered in the app.

Are you got success in this?

rakshitbharat, 2024-10-04:
I can confirm this problem is back in Oct 2024