innoveit/react-native-ble-manager #1142: checkState not working on ios

dejawho, 2023-12-18:
Seems that checkState is not working on ios, something like

     BleManager.checkState().then((state) => {
                console.log('state ' + state);
            }).catch(error => {
                console.log(' error ' + error)
            });

doesn't print the state nor the error. Is this expected? how can I check the state of the bt on ios in this case?

 - Device: [e.g. iPhone 15 plus]
 - react-native-ble-manager version: 11.0.6
 - react-native version: 0.72.6

marcosinigaglia, 2023-12-21:
Is working.

closed 2023-12-21 by marcosinigaglia

dejawho, 2023-12-21:
> Is working.

what do you mean? that it is working for you or that it is supposed to work like this?

marcosinigaglia, 2023-12-21:
The function is working and is printing the state.

dejawho, 2023-12-21:
> The function is working and is printing the state.

mmm, what could be the problem on my side? There must be some difference with the native code, this works in android but not on ios on my side (both on simulator or real device)

vhakulinen, 2023-12-22:
I have the same issue on ios, the promise doesn't resolve.

vhakulinen, 2023-12-22:
I have another issue in a similar vein: if I have to concurrent reads to same characteristic, only the other one will resolve.