innoveit/react-native-ble-manager #1147: BluetoothGatt.writeDescriptor missing?

matheusjardimb, 2024-01-02:
I'm trying to replicate an Android native app using ReactNative and react-native-ble-manager.

The code I'm replicating uses `android.bluetooth.BluetoothGatt.writeDescriptor()`, which I couldn't find any respective method here in the docs. Am I missing something? I'm using v11.0.8.

Only found one reference to such method at:
https://github.com/search?q=repo%3Ainnoveit%2Freact-native-ble-manager%20writeDescriptor&type=code

Besides that, the app is working just fine. Thanks for your help!

marcosinigaglia, 2024-01-04:
Hi @matheusjardimb , the writeDescriptor method now is only used to enable the notification.

marcosinigaglia, 2024-01-04:
Hi @matheusjardimb , can you test the [branch](https://github.com/innoveit/react-native-ble-manager/tree/feat/write-descriptor) to see if it's working?

matheusjardimb, 2024-01-04:
Sure! I'll check it out and give you a feedback asap

matheusjardimb, 2024-01-08:
@marcosinigaglia, I just tested it and it seems to be working just fine. Thanks

closed 2024-01-11 by marcosinigaglia