innoveit/react-native-ble-manager #1152: On devices with Android>12, BleManagerDiscoverPeripheral cannot be triggered

looooool11111, 2024-01-11:
"react-native-ble-manager": "^11.1.0",
    "react-native": "0.72.7",
    I found that on devices with Android version greater than 12, the app cannot listen to BleManagerDiscoverPeripheral. Here is my code

     useEffect(() => {
        console.log('needInit-start',needInit)
        try {
            BleManager.start({showAlert: false})
            .then(() => console.log('BleManager started.'))
            .catch((error: any) =>
                console.log('BeManager could not be started.', error),
            );
        } catch (error) {
            console.log('unexpected error starting BleManager.', error);
            return;
        }
        const listeners = [
            bleManagerEmitter.addListener('BleManagerDiscoverPeripheral',handleDiscoverPeripheral), 
            bleManagerEmitter.addListener('BleManagerStopScan', handleStopScan),
            bleManagerEmitter.addListener('BleManagerDidUpdateState', handleDidUpdateState),
            bleManagerEmitter.addListener('BleManagerDisconnectPeripheral', handleDisconnectedPeripheral), 
bleManagerEmitter.addListener(BleEventType.BleManagerDidUpdateValueForCharacteristic,handleUpdateValueForCharacteristic), 
            bleManagerEmitter.addListener('BleManagerConnectPeripheral',handleConnectPeripheral)
        ];
        handleAndroidPermissions(); 
        return () => {
            for (const listener of listeners) {
            listener.remove();
            }
        };
    }, [needInit]);
    const handleDiscoverPeripheral = (peripheral: Peripheral) => {
        console.log(peripheral,'peripheral');
        if(peripheral.name && peripheral.name.indexOf("DM") == -1){
            return;
        }
        if (!peripheral.name && !peripheral.localName) {
            return
        }
        if (!peripheral.name) {
            peripheral.name = 'NO NAME';
        }
        setPeripherals(map => {
            return new Map(map.set(peripheral.id, peripheral));
        });
    };
    
    console.log(peripheral,'peripheral');Never triggered

marcosinigaglia, 2024-01-11:
Check the permissions.

closed 2024-01-11 by marcosinigaglia

looooool11111, 2024-01-11:
> Check the permissions.

<manifest xmlns:android="http://schemas.android.com/apk/res/android">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.CAMERA" />
    <application
      android:name=".MainApplication"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:roundIcon="@mipmap/ic_launcher_round"
      android:allowBackup="false"
      android:theme="@style/AppTheme">
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize|uiMode"
        android:launchMode="singleTask"
        android:windowSoftInputMode="adjustResize"
        android:screenOrientation="landscape"
        android:exported="true">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
    </application>
</manifest>

looooool11111, 2024-01-11:
> > Check the permissions.
> 
> ```
> <uses-permission android:name="android.permission.INTERNET" />
> <uses-permission android:name="android.permission.BLUETOOTH" />
> <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
> <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
> <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
> <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
> <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
> <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
> <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
> <uses-permission android:name="android.permission.CAMERA" />
> <application
>   android:name=".MainApplication"
>   android:label="@string/app_name"
>   android:icon="@mipmap/ic_launcher"
>   android:roundIcon="@mipmap/ic_launcher_round"
>   android:allowBackup="false"
>   android:theme="@style/AppTheme">
>   <activity
>     android:name=".MainActivity"
>     android:label="@string/app_name"
>     android:configChanges="keyboard|keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize|uiMode"
>     android:launchMode="singleTask"
>     android:windowSoftInputMode="adjustResize"
>     android:screenOrientation="landscape"
>     android:exported="true">
>     <intent-filter>
>         <action android:name="android.intent.action.MAIN" />
>         <category android:name="android.intent.category.LAUNCHER" />
>     </intent-filter>
>   </activity>
> </application>
> ```
and
[image]