innoveit/react-native-ble-manager #1153: Calling scan method is causing infinite loop

alexismoreau, 2024-01-11:
Calling 
`BleManager.scan([], 0, true)`
is causing infinite loop on React Native : 

`WARN  Please report Excessive number of pending callbacks: 501. Some pending callbacks that might have leaked by never being called from native code: {"6761":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6766":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6771":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6776":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6781":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6786":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6791":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6796":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6801":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6806":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6811":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6816":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6821":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6826":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6831":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6836":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6841":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6846":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6851":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6856":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6915":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6920":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6925":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6931":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"6937":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7125":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7130":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7135":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7140":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7145":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7150":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7155":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7160":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7165":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7170":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7175":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7180":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7185":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7190":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7195":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7200":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7205":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7210":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7397":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7402":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7407":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7412":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7417":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7422":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"7427":{"module":"NativeAnimatedModule","method":"startAnimatingNode"},"...(truncated keys)...":451}`

I find this issue on both iOS and Android.

Do you know what I'm doing wrong ?

Thanks

lucaswitch, 2024-01-16:
Can you provide the full code for this particular screen?

alexismoreau, 2024-01-19:
Sorry, this was caused by a function editing the react state, linked to the event listener.

closed 2024-01-19 by alexismoreau