innoveit/react-native-ble-manager #1166: writeWithoutrepose issue in iOS

PrudhviBanki, 2024-02-08:
**chunk data :**[108, 0, 108, 109, 97, 99, 80, 114, 111, 99, 101, 115, 115, 83, 104, 111, 114, 116, 82, 101, 116, 114, 121, 70, 97, 105, 108, 0, 108, 109, 97, 99, 80, 114, 111, 99, 101, 115, 115, 84, 120, 67, 111, 109, 112, 108, 101, 116, 101, 0, 108, 109, 97, 99, 80, 114, 111, 99, 101, 115, 115, 67, 111, 108, 108, 105, 115, 105, 111, 110, 0, 108, 109, 97, 99, 68, 105, 115, 99, 97, 114, 100, 70, 114, 97, 109, 101, 69, 120, 99, 104, 97, 110, 103, 101, 83, 101, 113, 117, 101, 110, 99, 101, 0, 108, 109, 97, 99, 80, 114, 111, 99, 101, 115, 115, 84, 120, 83, 117, 99, 99, 101, 115, 115, 0, 108, 109, 97, 99, 80, 114, 111, 99, 101, 115, 115, 84, 120, 111, 112, 81, 67, 111, 109, 112, 108, 101, 116, 101, 0, 112, 112, 46, 99, 0, 0, 94, 116, 56, 64, 94, 116, 56, 64, 94, 116, 56, 64, 94, 116, 56, 64, 92, 116, 56, 64, 104, 116, 56, 64, 75, 116, 56, 64, 67, 116, 56, 64, 56, 116, 56, 64, 252, 116, 56, 64, 252, 116, 56, 64, 252, 116, 56, 64, 252, 116, 56, 64, 148, 116, 56, 64, 211, 116, 56, 64, 7, 117, 56, 64, 116, 116, 56, 64, 127, 116, 56, 64, 136, 116, 56, 64, 4, 117, 56, 64, 4, 117, 56, 64, 4, 117, 56, 64, 159, 116, 56, 64, 170, 116, 56, 64]

**code:**
for loop with diff chunks sample provided in above

const dataBytes=Buffer.from(chunk, 'base64').toJSON().data;
                console.log("dataBytes", dataBytes);
                // console.log("length", dataBytes.length);//336
            
                BleManager.writeWithoutResponse(peripheralId,  "00006720-1809-2101-1357-65626173746F","00006722-1809-2101-1357-65626173746F", dataBytes,20,2)
                .then(() => {
                    console.log('Write operation successful');
                    this.updateCounterData(index,totalChunks,peripheralId)
    
                })
                .catch(error => {
                    console.error('Error during write operation:', error);
                    // Handle error
                });

**xcode log :**

[TOO BIG formatValueCalls 201 exceeded limit of 200],[CoreBluetooth] WARNING: Characteristic <CBCharacteristic: 0x28792a820, UUID = 00006722-1809-2101-1357-65626173746F, properties = 0x88, value = (null), notifying = NO> does not specify the "Write Without Response" property - ignoring response-less write
2024-02-08 10:41:13.855006+0530 

can you provide me solution i'm getting always   console.log('Write operation successful'); but it is not working

PrudhviBanki, 2024-02-08:
> **chunk data :**[108, 0, 108, 109, 97, 99, 80, 114, 111, 99, 101, 115, 115, 83, 104, 111, 114, 116, 82, 101, 116, 114, 121, 70, 97, 105, 108, 0, 108, 109, 97, 99, 80, 114, 111, 99, 101, 115, 115, 84, 120, 67, 111, 109, 112, 108, 101, 116, 101, 0, 108, 109, 97, 99, 80, 114, 111, 99, 101, 115, 115, 67, 111, 108, 108, 105, 115, 105, 111, 110, 0, 108, 109, 97, 99, 68, 105, 115, 99, 97, 114, 100, 70, 114, 97, 109, 101, 69, 120, 99, 104, 97, 110, 103, 101, 83, 101, 113, 117, 101, 110, 99, 101, 0, 108, 109, 97, 99, 80, 114, 111, 99, 101, 115, 115, 84, 120, 83, 117, 99, 99, 101, 115, 115, 0, 108, 109, 97, 99, 80, 114, 111, 99, 101, 115, 115, 84, 120, 111, 112, 81, 67, 111, 109, 112, 108, 101, 116, 101, 0, 112, 112, 46, 99, 0, 0, 94, 116, 56, 64, 94, 116, 56, 64, 94, 116, 56, 64, 94, 116, 56, 64, 92, 116, 56, 64, 104, 116, 56, 64, 75, 116, 56, 64, 67, 116, 56, 64, 56, 116, 56, 64, 252, 116, 56, 64, 252, 116, 56, 64, 252, 116, 56, 64, 252, 116, 56, 64, 148, 116, 56, 64, 211, 116, 56, 64, 7, 117, 56, 64, 116, 116, 56, 64, 127, 116, 56, 64, 136, 116, 56, 64, 4, 117, 56, 64, 4, 117, 56, 64, 4, 117, 56, 64, 159, 116, 56, 64, 170, 116, 56, 64]
> 
> **code:** for loop with diff chunks sample provided in above
> 
> const dataBytes=Buffer.from(chunk, 'base64').toJSON().data; console.log("dataBytes", dataBytes); // console.log("length", dataBytes.length);//336
> 
> ```
>             BleManager.writeWithoutResponse(peripheralId,  "0000xxx-1809-2101-1357-65626173746F","0000xxxx-1809-2101-1357-65626173746F", dataBytes,20,2)
>             .then(() => {
>                 console.log('Write operation successful');
>                 this.updateCounterData(index,totalChunks,peripheralId)
> 
>             })
>             .catch(error => {
>                 console.error('Error during write operation:', error);
>                 // Handle error
>             });
> ```
> 
> **xcode log :**
> 
> [TOO BIG formatValueCalls 201 exceeded limit of 200],[CoreBluetooth] WARNING: Characteristic <CBCharacteristic: 0x28792a820, UUID = 00006722-1809-2101-1357-65626173746F, properties = 0x88, value = (null), notifying = NO> does not specify the "Write Without Response" property - ignoring response-less write 2024-02-08 10:41:13.855006+0530
> 
> can you provide me solution i'm getting always console.log('Write operation successful'); but it is not working in iOS

android getting E (4324007) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT

marcosinigaglia, 2024-02-08:
It appears that the feature does not support writeWithoutResponse

PrudhviBanki, 2024-02-09:
Any solution for this?

marcosinigaglia, 2024-02-09:
You have to check your device documentation.

closed 2024-02-09 by marcosinigaglia

longphi97, 2024-07-10:
can u fix it?
@PrudhviBanki