innoveit/react-native-ble-manager #1182: Bluetooth BleManagerDidUpdateValueForCharacteristic event emits Service UUIDs and Characteristics UUIDs different between iOS and Android.

lucaswitch, 2024-03-08:
**Describe the bug**
When trying to monitor some characteristics the emitted value on iOS the current characteristic/uuid are not in the same format that is given to  "startNotification" method and the "BleManagerDidUpdateValueForCharacteristic" event.

**To Reproduce**
Lets try to monitor the device with characteristics and services in the following format:
```js

const HEART_DEVICE_SERVICE_UUID = '0000180D-0000-1000-8000-00805F9B34FB';
const HEART_DEVICE_SERVICE_CHARACTERISTIC_UUID =
  '00002a37-0000-1000-8000-00805f9b34fb';

  let listener = BleManagerEmitter.addListener(
      'BleManagerDidUpdateValueForCharacteristic',
      onCharacteristicChange,
    );
// The full format of UUID was given.
    BleManager.startNotification(id, HEART_DEVICE_SERVICE_UUID, HEART_DEVICE_SERVICE_CHARACTERISTIC_UUID)
      .then(function () {
        onSuccess();
      })
      .catch(function (err) {
        console.error(err);
        onError(err);
      });
```

But afterwards on characteristic change we receive the above UUIDs on the short format:

```js
 function onCharacteristicChange({
      value,
      peripheral,
      characteristic,
      service,
    }) {
      const isAllowed = peripheral === id;
      // On Apple Devices:
      // Service UUID equals to: '2a37'
      // Characteristic UUID equals to: '180D'
     // On Android devices the UUID is in a completely different format.

      if (isAllowed) {
        onNotify(value);
      }
    }

    let listener = BleManagerEmitter.addListener(
      'BleManagerDidUpdateValueForCharacteristic',
      onCharacteristicChange,
    );
```
### Explanation 

The real problem relies on the undefined behavior on BleManagerDidUpdateValueForCharacteristic event between platforms.
The provided service UUID and characteristic UUID should be provided back into the BleManagerDidUpdateValueForCharacteristic event callback.

I can confirm that:
 - On Android always the full format is given to the BleManagerDidUpdateValueForCharacteristic callback.
 - On iOS/IpadOs only the short format is given to the BleManagerDidUpdateValueForCharacteristic callback.
 
```

## Expected behavior

The expected behavior is the characteristics UUIDs and services UUIDs provided on "startNotification" be exactly in the same format provided by the BleManagerDidUpdateValueForCharacteristic callback.

The exact react-native-ble-manager version: 11.3.2

## Additional context
Thank you for the awesome work and effort and continuous integration into this library.

lotalogic, 2024-03-14:
Hi, 
I have some kind of similar issue in the sense that my application works fine on android and not on iOs.
Do you have any idea why : [https://github.com/innoveit/react-native-ble-manager/discussions/1184](https://github.com/innoveit/react-native-ble-manager/discussions/1184)

lucaswitch, 2024-03-15:
> Hi, I have some kind of similar issue in the sense that my application works fine on android and not on iOs. Do you have any idea why : #1184

I'm guessing it's not related to this issue, this issue happens on transmission between characteristics

lotalogic, 2024-03-15:
Ok thanks

lucaswitch, 2024-06-12:
For anyone that could have this issue it can be solved on react-native app level.
In order to support many known short characteristics/services uuids we can do as following:

```js 

function onCharacteristicChange({
      value,
      peripheral,
      characteristic,
      service,
    }) {
      if (peripheral === id) {
        let isSameCharacteristic = false;
        let isSameService = false;

        // Characteristic
        if (characteristic.length === 4) {
          isSameCharacteristic =
            characteristic.toLowerCase() === shortCharacteristicId;
        } else {
          isSameCharacteristic =
            characteristic.toLowerCase() === characteristicId;
        }

        // Service
        if (service.length === 4) {
          isSameService = service.toLowerCase() === shortServiceId;
        } else {
          isSameService = service.toLowerCase() === serviceId;
        }

        if (isSameService && isSameCharacteristic) {
           console.log('it works');
        }
      }
 }

  let listener = BleManagerEmitter.addListener(
    "BleManagerDidUpdateValueForCharacteristic",
    onCharacteristicChange
  );
`

closed 2024-06-12 by lucaswitch