innoveit/react-native-ble-manager #1183: requestMTU never resolves

kurogalaxy, 2024-03-14:
**Describe the bug**
After connecting I call the requestMTU method and it never resolves so the app won't continue with the flow.

**To Reproduce**
Steps to reproduce the behavior:
1. Ask all permissions/Check permissions
2. Scan devices
3. Connect to device
4. Call requestMTU with different sizes (128, 256, 512)

**Expected behavior**
requestMTU should resolve with either an error or a number.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: Samsung
 - OS: Android 14 & Android 13
 - react-native-ble-manager version: 11.3.2
 - react-native version: 0.72.6

**Additional context**
On my Android 13 phone it usually works but there are some times that it won't resolve either.

aidansauce, 2024-03-18:
I've seen the same thing today when porting some code that was written against a much older version (7.4.1). However, when I stepped through the code in Android Studio I couldn't recreate it.

That's led me to add a 500ms sleep between the connect promise resolving and calling requestMTU which seems to be enough

I added some extra logging to try and find the issue and I'm reasonably sure that in Peripheral.java `gatt.requestMtu()` is being called, but `onMtuChanged` is not so this may be a behaviour change in Android (or maybe Samsung?) since my code was last used. 

- Samsung Android 14
- Version 11.3.2
- RN 0.72.12 
- Connecting to a bespoke device

marcosinigaglia, 2024-06-18:
This issue is stale because it has been open 180 days with no activity. Remove stale label or comment or this will be closed in 7 days.

thomasttvo, 2024-06-18:
I ran into this for any value smaller than `23`, but then I realized it's the standard BLE protocol that the MTU is 23 or higher, so may that explained it. I think `react-native-ble-manager` should add a line in `requestMTU` that ensures the value is higher or equal 23
```
requestMTU(peripheralId: string, mtu: number) {
   mtu = Math.max(mtu, 23);

```

closed 2024-09-25 by marcosinigaglia