innoveit/react-native-ble-manager #1186: BLE manager library not supporting API level 34.

NikitaBH-Druck, 2024-03-21:
Steps to reproduce
In Android 14(SDK 34) scan function is not working

Expected behavior
it should work as previous android versions

Actual behavior
App crash when call the BLE Scan 

Environment
React Native version: 0.73.6.
React Native platform + platform version: Android 14.0 

[image]

Sofrosine, 2024-03-24:
i also have this issue

lucaswitch, 2024-03-26:
That should have a easy fix.

NikitaBH-Druck, 2024-03-27:
I have made changes as per ([https://github.com/innoveit/react-native-ble-manager/pull/1188/files/e60195660ec1faa2a65eb9b35710abcdba853b91..9924eb4aa0f4691ad6c85d448c6c66ff67266d53)]) and tested on real device with android 14  but getting same error as attached screenshot.

lucaswitch, 2024-03-27:
> I have made changes as per ([https://github.com/[/pull/1188](https://github.com/innoveit/react-native-ble-manager/pull/1188)/files/e60195660ec1faa2a65eb9b35710abcdba853b91..[9924eb4](https://github.com/innoveit/react-native-ble-manager/commit/9924eb4aa0f4691ad6c85d448c6c66ff67266d53))]) and tested on real device with android 14 but getting same error as attached screenshot.
> 
> We will need urgent support for this as users are using the Application for Android 14 also.

Added a change again, can you gently test it again?

NikitaBH-Druck, 2024-03-27:
I have tested but not working, Does we also need to change TargetSDK version in build.gradle of react-native-Ble-manager from 33 to 34 for supports android 14...

lucaswitch, 2024-04-01:
Just bought a Android 14 device. And can confirm https://github.com/innoveit/react-native-ble-manager/pull/1188/commits/dd29fa7bc5ae5d39a3346706c28143af53eae570 solves it.

closed 2024-04-01 by NikitaBH-Druck