innoveit/react-native-ble-manager #1196: [iOS] [checkState] Always returns 'unknown'

joenini, 2024-04-08:
**Describe the bug**
On iOS, using bleManager.checkState() always returns 'unknown'. I tried with Bluetooth permissions granted/ not granted, and with Bluetooth on/ off.

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone8/11]
 - OS: [e.g. iOS16/17]
 - react-native-ble-manager version: 11.3.2
 - react-native version: 0.72.6

**Additional context**
However, the listener for BleManagerDidUpdateState is receiving the correct state ['unauthorized', 'off', 'on']. 
Also, on Android bleManager.checkState() returns the correct state.

lucaswitch, 2024-04-08:
Make sure to wrap it on a "await" call. I'm also using 11.3.2 and never had this problem.
[image]

joenini, 2024-04-08:
I do have await, same as in your screenshot.

hisstoryxx, 2024-04-25:
I have same ISSUE!! 
using 11.4.0 version in iOS

hisstoryxx, 2024-04-26:
`await BleManager.checkState();`
I think this function doesn't guarantee completion.
so need to check the state of **BleManagerDidUpdateState**

 await BleManager.start();

        const listener = BleManager.addListener(
          BleEventType.BleManagerDidUpdateState,
          event => {
            console.log(
              `${new Date().toISOString()} =============================================`,
            );
            console.log(event);

            BleManager.removeAllListeners(
              BleEventType.BleManagerDidUpdateState,
            );
          },
        );

lucaswitch, 2024-05-10:
Make sure to call functions just after start method cause ble manager needs to have CBCentralManager instance ready to do proper work, see: https://github.com/innoveit/react-native-ble-manager/blob/master/ios/BleManager.swift
Probably not a library issue.

There's the behaviors:

**checkState(imperative): Gives the current state even if it's not ready will report unknown asap.**

**BleManagerDidUpdateState(Event driven): As soon CBCentralManager is ready with start to report changes**

closed 2024-08-17 by marcosinigaglia