innoveit/react-native-ble-manager #1197: android14 : Illegal callback invocation from native module. This callback type only permits a single invocation from native code.

stonehiy, 2024-04-10:
**Before open an issue**

**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Phone's bluetooth is close
2. BleManager.enableBluetooth().catch(() => {});

**Expected behavior**

**Screenshots**

**Smartphone (please complete the following information):**
 - Device: [e.g.OnePlus 11 ]
 - OS: [e.g.andorid14]
 - react-native-ble-manager version: 11.5.0
 - react-native version: 0.72

**Additional context**

 Caused by: java.lang.RuntimeException: Illegal callback invocation from native module. This callback type only permits a single invocation from native code.
                                                                                                    	at com.facebook.react.bridge.CallbackImpl.invoke(CallbackImpl.java:26)
                                                                                                    	at it.innove.BleManager$1.onActivityResult(BleManager.java:95)
                                                                                                    	at com.facebook.react.bridge.ReactContext.onActivityResult(ReactContext.java:338)
                                                                                                    	at com.facebook.react.ReactInstanceManager.onActivityResult(ReactInstanceManager.java:821)
                                                                                                    	at com.facebook.react.ReactDelegate.onActivityResult(ReactDelegate.java:90)
                                                                                                    	at com.facebook.react.ReactActivityDelegate.onActivityResult(ReactActivityDelegate.java:133)
                                                                                                    	at com.facebook.react.ReactActivity.onActivityResult(ReactActivity.java:70)
                                                                                                    	at android.app.Activity.dispatchActivityResult(Activity.java:9116)
                                                                                                    	at android.app.ActivityThread.deliverResults(ActivityThread.java:5878)
                                                                                                    	at android.app.ActivityThread.handleSendResult(ActivityThread.java:5924) 
                                                                                                    	at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:67) 
                                                                                                    	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45) 
                                                                                                    	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:144) 
                                                                                                    	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:101) 
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2630) 
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106) 
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:257) 
                                                                                                    	at android.os.Looper.loop(Looper.java:368) 
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8826) 
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method) 
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:572) 
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1049) 

```
private final ActivityEventListener mActivityEventListener = new BaseActivityEventListener() {

        @Override
        public void onActivityResult(Activity activity, int requestCode, int resultCode, Intent intent) {
            Log.d(LOG_TAG, "onActivityResult");
            if (requestCode == ENABLE_REQUEST && enableBluetoothCallback != null) {
                if (resultCode == RESULT_OK) {
                    enableBluetoothCallback.invoke();
                } else {
                    enableBluetoothCallback.invoke("User refused to enable"); // it.innove.BleManagerS1.onActivityResult(BleManager.java:95)
                }
                enableBluetoothCallback = null;
            }
        }

    };
```

Super-Bin, 2024-04-16:
same issue

marcosinigaglia, 2024-04-16:
I tested with no issue.

Joaoo1, 2024-04-21:
same issue here too. Android 12.

lucaswitch, 2024-04-24:
> This callback type only permits a single invocation from native code.

Can you provide a minimal repro for this?
Also make sure to call enableBluetooth() only once per time.

marcosinigaglia, 2024-07-24:
This issue is stale because it has been open 180 days with no activity. Remove stale label or comment or this will be closed in 7 days.

closed 2024-08-01 by marcosinigaglia

andhikaalgo, 2024-12-19:
same issue