innoveit/react-native-ble-manager #1204: createBond never resolves

Bartosz-L, 2024-04-11:
**Describe the bug**
createBond(id) never resolves. The bonding did happen correctly, as devices are bonded, but the promise is not resolving.

**To Reproduce**
Steps to reproduce the behavior:
1. Ask all permissions/Check permissions
2. Scan devices
3. Connect to device
4. Call requestMTU
5. Call createBond() with given id
6. Clicked Pair on the native popup
7. Bonding is happening but promise not resolving

**Expected behavior**
createBond Promise should resolve

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: Samsung S23
 - OS: Android 14
 - react-native-ble-manager version: 11.5.0
 - react-native version: 0.72.7

**Additional context**
same goes for removeBond, any other method resolves normally but not those two

FlorinPopaPersonal, 2024-04-19:
I had a similar issue and I managed to fix it by changing 
`if (Build.VERSION.SDK_INT >= UPSIDE_DOWN_CAKE){
            // Google in 2023 decides that flag RECEIVER_NOT_EXPORTED or RECEIVER_EXPORTED should be explicit set SDK 34(UPSIDE_DOWN_CAKE) on registering receivers.
            // Also the export flags are available on Android 8 and higher, should be used with caution so that don't break compability with that devices.
            context.registerReceiver(mReceiver, filter, Context.RECEIVER_NOT_EXPORTED);
            context.registerReceiver(mReceiver, intentFilter, Context.RECEIVER_NOT_EXPORTED);
        }`
to
`if (Build.VERSION.SDK_INT >= UPSIDE_DOWN_CAKE){
            // Google in 2023 decides that flag RECEIVER_NOT_EXPORTED or RECEIVER_EXPORTED should be explicit set SDK 34(UPSIDE_DOWN_CAKE) on registering receivers.
            // Also the export flags are available on Android 8 and higher, should be used with caution so that don't break compability with that devices.
            context.registerReceiver(mReceiver, filter, Context.RECEIVER_EXPORTED);
            context.registerReceiver(mReceiver, intentFilter, Context.RECEIVER_EXPORTED);
        }`

lucaswitch, 2024-04-20:
This issue was introduced after version [11.4.0] which adds support to Android 14, (https://github.com/innoveit/react-native-ble-manager/releases/tag/v.11.4.0) and should be fixed with that pull request.

closed 2024-04-22 by marcosinigaglia