innoveit/react-native-ble-manager #1206: android app can not discover any device by scaning function

yinghuochongfire, 2024-04-18:
**Describe the bug**
in most case,android app can work well.
but, sometimes, android app can not discover any device by scaning device.
when i try to restart my phone, and run the app again, the app works well again, it can discover device quickly.
[scan device.txt](https://github.com/innoveit/react-native-ble-manager/files/15027171/scan.device.txt)

when i exit my app and restart my app again,This situation is often repeated.

**Smartphone (please complete the following information):**
 - Device: honor
 - OS: magic OS 7.2
 - react-native-ble-manager version: 11.0.6
 - react-native version:0.73.0

yinghuochongfire, 2024-04-18:
now ,i find a new case.
When I manually enable Bluetooth on my phone before the APP runs, the APP does not return any device when scanning Bluetooth devices. But when the APP run before, I manually disable my phone bluetooth,and when the APP runs, through the code "BleManager. EnableBluetooth ()" enable the bluetooth, and then scan, can return devices. 
Repeat many times, the phenomenon is the same

marcosinigaglia, 2024-04-19:
So probably is a device problem.

lucaswitch, 2024-04-24:
**Don't think it's a library bug.** 
But quick looking at your implementation i could notice something.
**Make sure to register the events callback when the bluetooth is turned on, permission ready and when callbacks needed**, make sure to unregistering it when it's not needed anymore.

For that scanning you should do:
1) Make sure bluetooth is on
2) Make sure permissions is granted
3) Start scan, listen to discovered devices.

And afterwards when scan is not running

1) Empty discovered devices
2) Unregister scan discovery callback.

mohammadrajabraza, 2024-05-07:
I began with the example provided in the library and encountered a similar problem as @yinghuochongfire . The scan function doesn't seem to work until I attempt it with companionScan. Once a device is discovered, I can close the app and then try scan again, and it works.

Smartphone (please complete the following information):

Device: MI 10T
react-native-ble-manager version: 11.5.3
react-native version:0.73.2

jeslo, 2024-05-22:
I have a related issue in Android. When launching the app, it asks for permission, and I grant it, but the scan is not working. I checked the phone settings, and the permissions are granted. When I toggle the permission off and then on again from my physical device's settings, the scan works perfectly. So, the permission needs to be granted from the physical device settings.

theflyinhawaiian, 2024-05-29:
> I began with the example provided in the library and encountered a similar problem as @yinghuochongfire . The scan function doesn't seem to work until I attempt it with companionScan. Once a device [...]

Thank you for this!! 

You'll also need to add:

```<uses-feature android:name="android.software.companion_device_setup"/>```

to your manifest if you haven't, in order to get `companionScan` to run.

renochew, 2024-06-03:
Just want to chime in the exact situation as described by @yinghuochongfire happen to me today. I am scratch my head as handleDiscoverPeripheral just can't catch any result from a scan, whatever I try, but once I did a successful companion search, subsequent scan will be successful.  

And I can't even reproduce the original problem anymore, as if the device can't "unseen" the device. It might be a good idea to test it with another phone, I don't have one at the moment. I don't think the original problem is solved, something still not right before we did the companion search.

closed 2024-09-09 by marcosinigaglia