innoveit/react-native-ble-manager #1219: Bluetooth permission on iOS is asked on App launch

shettayyy, 2024-05-09:
**Describe the bug**

Bluetooth permission dialog opens up as soon as I launch the app. i would want the users to see it on the screen where they are using this library rather than showing on app launch

**To Reproduce**
I am using expo

1. Install the package
2. Create an expo development build
3. Open the app
4. You should see the permission dialogue pop-up

**Expected behavior**
Show the permission dialog where the manager is used/started.

**Smartphone (please complete the following information):**
 - Device: iPhone 14 Pro Max
 - OS: iOS 17.4.1
 - react-native-ble-manager version: 11.5.3
 - react-native version: 0.73.6

lucaswitch, 2024-05-10:
I guess that's the default behavior for iOS for any device using bluetooth. You can customize the message on the infoplist.

shettayyy, 2024-05-10:
@lucaswitch So, there isn't a way to show this dialog where this package is used? There is a possibility of users dismissing the dialog mistakenly. I would prefer to show my dialog component before showing the system dialog for Bluetooth permissions. But, at present, I am not sure if it's possible if that's the default behavior for iOS.

shettayyy, 2024-05-10:
@lucaswitch https://stackoverflow.com/questions/29456654/bluetooth-le-ask-permission

From this post, it seems like it is possible to ask permission when we actually would like to use the feature.

lucaswitch, 2024-05-12:
Got it. For doing that is pretty straightforward. 
Watch the bluetooth adapter state and make that dialog. If the state is not but "on" then user has to get that dialog. Hope it helps!

shettayyy, 2024-05-14:
@lucaswitch I don't think my app gets the opportunity to check for the state. As explained under the description, the pop-up appears as soon as the app opens. I have not written any code to start the Ble manager at the app launch. It appears without calling the `Blemanager.start()`

So, install the package on Expo (Expo development build) and start the app on an actual device. It asks for Bluetooth permissions. I wanna ask it on the page where the Bluetooth feature is being used.

jeslo, 2024-05-22:
I have a related issue in Android. When launching the app, it asks for permission, and I grant it, but the scan is not working. I checked the phone settings, and the permissions are granted. When I toggle the permission off and then on again from my physical device's settings, the scan works perfectly. So, the permission needs to be granted from the physical device settings.

Device: MI 11X
react-native-ble-manager version: 11.5.3
react-native version:0.74.1

marcosinigaglia, 2024-08-21:
This issue is stale because it has been open 180 days with no activity. Remove stale label or comment or this will be closed in 7 days.

shettayyy, 2024-08-21:
I think this is something to look into as there is quite a possibility of users mistakenly skipping the prompt shown as soon as the app is opened. I don't think it should be closed unless someone reproduces this issue and the prompt is not shown at the beginning in iOS. I humbly request the ability to ask for Bluetooth permission when we use it as opposed to asking when the app opens.

closed 2024-11-27 by marcosinigaglia