innoveit/react-native-ble-manager #1221: Issue with large data transmission

varun761, 2024-05-15:
The BleManagerDidUpdateValueForCharacteristic didn't trigger on large data transmission. If many bytes are sent from ble device to the app. It doesn't detect the data change and doesn't trigger it.

Expected behaviour:
It needs to be triggered what ever the data length.

varun761, 2024-05-15:
Code

```
import {
  View,
  Text,
  TouchableOpacity,
  Platform,
  FlatList,
  Dimensions,
  NativeEventEmitter,
  NativeModules,
  SafeAreaView,
} from "react-native";
import {
  PERMISSIONS,
  RESULTS,
  checkMultiple,
  requestMultiple,
} from "react-native-permissions";
import React, {useState, useEffect, useCallback} from "react";
import BleManager from "react-native-ble-manager";
import {shallowEqual, useDispatch, useSelector} from "react-redux";
import {Buffer} from "buffer";

import styles from "./style";
import TopBar from "../../components/TopBar";
import { addTempData, addUpdateItem, selectorConnectedDevices, selectorDeviceConnected, updateScanningStatus, updateSensorData, updatedBluetoothStatus } from "../../redux/slices/deviceSlice";
import SectionHeading from "../../components/SectionHeading";
import { useFocusEffect } from "@react-navigation/native";

const {height} = Dimensions.get("window");
const navHeight = Platform.OS === "android" ? 55 : 85;
const screenArea = height - navHeight;

const BleManagerModule = NativeModules.BleManager;
const bleManagerEmitter = new NativeEventEmitter(BleManagerModule);

const serviceUUIDList = ["8de5e532-7565-4c8b-b333-5866350be21f"];

const ListBluetoothDevices = () => {

  const dispatch = useDispatch()
  
  const deviceState = useSelector(state => state.device);

  const discoveredDevices = selectorConnectedDevices(deviceState);
  const isDeviceConnected = selectorDeviceConnected(deviceState);

  // console.log('---isDeviceConnected---- :', isDeviceConnected)

  const isScanning = useSelector(
    state => state.device.scanningDevices === true,
  );

  const checkDeviceBlePermissions = async () => {
    return new Promise(async (resolve, reject) => {
      console.log("checking permissions");
      try {
        if (Platform.OS === "android") {
          const permissionsList = [
            PERMISSIONS.ANDROID.BLUETOOTH_SCAN,
            PERMISSIONS.ANDROID.BLUETOOTH_ADVERTISE,
            PERMISSIONS.ANDROID.BLUETOOTH_CONNECT,
          ];
          const permissions = await checkMultiple(permissionsList);
          const requestPermission = Object.keys(permissions).filter(
            el =>
              permissions[el] !== RESULTS.UNAVAILABLE &&
              permissions[el] !== RESULTS.GRANTED &&
[... 445 lines cut ...]
                style={{maxHeight: screenArea * 0.27}}
                data={discoveredDevices}
                renderItem={({item}) => (
                  <RenderBleDeviceLists
                    peripheral={item}
                  />
                )}
                keyExtractor={item => `${item.id}_discovered`}
              />
            </SafeAreaView>
          ) : (
            <Text style={{color: "#000", paddingHorizontal: 2}}>
              {isScanning
                ? "Scanning for devices ... "
                : "No Device Discovered"}
            </Text>
          )}
        </View>
      </View>
    </View>
  )
}

export default ListBluetoothDevices
```

varun761, 2024-05-15:
After requestMTU the data was received. But it suppose to receive data without requestMTU.

marcosinigaglia, 2024-05-15:
Is not a bug, is how BLE is designed in the OS

closed 2024-05-15 by marcosinigaglia

varun761, 2024-05-15:
Every time we need to call requestMTU while receive data? @marcosinigaglia

marcosinigaglia, 2024-05-15:
The default MTU on Android is 20, so after connecting it should be set to the size you need.

varun761, 2024-05-15:
And what is the maximum MTU we can set? I set it to 1000 and it never return any thing.

varun761, 2024-05-15:
@marcosinigaglia I am receiving 160 values in one transmission. It will transmit 2560 values total. Then what to set MTU and what will be approach ?

varun761, 2024-06-05:
@marcosinigaglia I hope this information will help

`06-06 00:40:03.941   941 22637 D r_submix: read_frames 21813390336, T_rec 454445.656250s, p_vs_o   15209us
06-06 00:40:04.004   942   965 D mtk.hal.bt@1.1-impl: BtHostDebugInfo
06-06 00:40:04.004   942   965 D [BT]    : mtk_bt_op: BT_VND_OP_SET_BTHOST
06-06 00:40:04.008  2096  2555 I bt_stack: [INFO:btsnoop.cc(342)] clear_l2cap_allowlist: Clearing acceptlist from l2cap channel. conn_handle=512 cid=4:4
06-06 00:40:04.009  2096  2555 I bt_stack: [INFO:gatt_attr.cc(338)] gatt_connect_cback: remove untrusted client status, bda=e7:01:a9:03:d3:8c
06-06 00:40:04.009  2096  2555 I bluetooth: system/bt/stack/gap/gap_ble.cc:298 client_connect_cback: No active GAP service found for peer:xx:xx:xx:xx:d3:8c callback:Disconnected
06-06 00:40:04.011  2096  2555 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1148 bta_gattc_conn_cback: Disconnected att_id:3 transport:le reason:GATT_CONN_TIMEOUT
06-06 00:40:04.011  2096  2555 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1148 bta_gattc_conn_cback: Disconnected att_id:4 transport:le reason:GATT_CONN_TIMEOUT
06-06 00:40:04.011  2096  2555 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1148 bta_gattc_conn_cback: Disconnected att_id:5 transport:le reason:GATT_CONN_TIMEOUT
06-06 00:40:04.011  2096  2555 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1148 bta_gattc_conn_cback: Disconnected att_id:6 transport:le reason:GATT_CONN_TIMEOUT
06-06 00:40:04.011  2096  2555 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1148 bta_gattc_conn_cback: Disconnected att_id:7 transport:le reason:GATT_CONN_TIMEOUT
06-06 00:40:04.011  2096  2555 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1148 bta_gattc_conn_cback: Disconnected att_id:8 transport:le reason:GATT_CONN_TIMEOUT
06-06 00:40:04.012  2096  2555 W bt_btm_sec: system/bt/stack/btm/btm_sec.cc:3849 btm_sec_disconnected: Got uncommon disconnection reason:Connection Timeout handle:0x0200
06-06 00:40:04.012  2096  2555 I btif_av : system/bt/btif/src/btif_av.cc:4184 btif_av_get_peer_sep: No active sink or source peer found
06-06 00:40:04.012  2096  2555 I bt_btif_a2dp_source: system/bt/btif/src/btif_a2dp_source.cc:850 btif_a2dp_source_on_idle: btif_a2dp_source_on_idle: state=STATE_OFF
06-06 00:40:04.012  2096  2555 I btif_av : system/bt/btif/src/btif_av.cc:4184 btif_av_get_peer_sep: No active sink or source peer found
06-06 00:40:04.012  2096  2555 I bt_btif_a2dp_source: system/bt/btif/src/btif_a2dp_source.cc:850 btif_a2dp_source_on_idle: btif_a2dp_source_on_idle: state=STATE_OFF
06-06 00:40:04.012   942   965 D mtk.hal.bt@1.1-impl: BtHostDebugInfo
06-06 00:40:04.012   942   965 D [BT]    : mtk_bt_op: BT_VND_OP_SET_BTHOST
06-06 00:40:04.015  2096 31976 D BluetoothMetrics: btclass1f00
06-06 00:40:04.015 27885 27925 D BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=E7:01:A9:03:D3:8C
06-06 00:40:04.016   942   965 D mtk.hal.bt@1.1-impl: BtHostDebugInfo
06-06 00:40:04.016   942   965 D [BT]    : mtk_bt_op: BT_VND_OP_SET_BTHOST
06-06 00:40:04.016 27885 27925 D RNBleManager: onConnectionStateChange to 0 on peripheral: E7:01:A9:03:D3:8C with status 0
06-06 00:40:04.018  2096 31976 D Checkin : publish the event [tag = MOT_BT event name = DISC]
06-06 00:40:04.020 27885 27885 D BluetoothGatt: cancelOpen() - device: E7:01:A9:03:D3:8C
06-06 00:40:04.024 27885 27885 D BluetoothGatt: close()
06-06 00:40:04.024 27885 27885 D BluetoothGatt: unregisterApp() - mClientIf=8
06-06 00:40:04.034 27885 27885 D RNBleManager: Peripheral event (BleManagerDisconnectPeripheral):E7:01:A9:03:D3:8C
06-06 00:40:04.035  2128  2223 W BluetoothEventManager: AclStateChangedHandler: activeDevice is null`

varun761, 2024-06-05:
The device disconnects on BLE_GATT_STATUS_SUCCESS   0x0000. code

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.api.v3.0.0%2Fgroup___b_l_e___g_a_t_t___s_t_a_t_u_s___c_o_d_e_s.html

itsaliraxa, 2024-07-11:
> Code
>
> ```
> import {
>   View,
>   Text,
>   TouchableOpacity,
>   Platform,
>   FlatList,
>   Dimensions,
>   NativeEventEmitter,
>   NativeModules,
>   SafeAreaView,
> } from "react-native";
> import {
>   PERMISSIONS, [...]

HI @varun761  I am not able get the values from this listener
 BleManagerDidUpdateValueForCharacteristic, can you please help me how I can get data?

varun761, 2024-07-15:
Increasing the data transmission time can help you out. If it's 5s right now make it 10 seconds or more. Processing of data depends upon the mobile device.

itsaliraxa, 2024-07-15:
Hi Varun! Can we do google meet?

Qdafengzi, 2025-05-13:
Has the problem been solved?